EtchedPixels/FUZIX #797: v65c816 init hangs/panic

yapinm, 2020-07-05:
Hi,
I'm trying to run Fuzix built for v65c816 platform, but i can't pass init. 
Here's what I've tried:

Case 1:
Fuzix on (atm latest) commit: af6bd7bd19dcd12206da7bf5070d979c6de8530e
locally built cc65: cc65 V2.18 - Git 17c55041
virtual 65816 from https://github.com/EtchedPixels/

What i did:
Makfile changed target: TARGET=v65c816
Did make and build-mini-filesystem disk0 256 65535 for disk0
Copied disk0 to v65c816 and did make and make os.
Result:
Started v65 and choosing bootdev 0 gives me this:
bootdev: 0
Mounting root fs (root_dev=0,ro): ??? h  Starting /init
and it just hangs in there.
v65 -t gives this log:
--- SNIP ---
A=FF04 X=0000 Y=0006 S=B10E D=FF00 B=00 P=35 E=0  00/40F3  JMP $A28C (@00A28C A0 08 4C ...)
A=FF04 X=0000 Y=0006 S=B10E D=FF00 B=00 P=35 E=0  00/A28C  LDY #$08
A=FF04 X=0000 Y=0008 S=B10E D=FF00 B=00 P=35 E=0  00/A28E  JMP $A087 (@00A087 48 18 98 ...)
A=FF04 X=0000 Y=0008 S=B10E D=FF00 B=00 P=35 E=0  00/A087  PHA
A=FF04 X=0000 Y=0008 S=B10D D=FF00 B=00 P=35 E=0  00/A088  CLC
A=FF04 X=0000 Y=0008 S=B10D D=FF00 B=00 P=34 E=0  00/A089  TYA
A=FF08 X=0000 Y=0008 S=B10D D=FF00 B=00 P=34 E=0  00/A08A  ADC $00 (@00FF00 05 B2 15 ...)
A=FF0D X=0000 Y=0008 S=B10D D=FF00 B=00 P=34 E=0  00/A08C  STA $00 (@00FF00 05 B2 15 ...)
A=FF0D X=0000 Y=0008 S=B10D D=FF00 B=00 P=34 E=0  00/A08E  BCC $02 ($00A092)
A=FF0D X=0000 Y=0008 S=B10D D=FF00 B=00 P=34 E=0  00/A092  PLA
A=FF04 X=0000 Y=0008 S=B10E D=FF00 B=00 P=34 E=0  00/A093  RTS
A=FF04 X=0000 Y=0008 S=B110 D=FF00 B=00 P=34 E=0  00/0658  JMP $0684 (@000684 9C C2 CC ...)
A=FF04 X=0000 Y=0008 S=B110 D=FF00 B=00 P=34 E=0  00/0684  STZ $CCC2 (@00CCC2 01 07 0A ...)
A=FF04 X=0000 Y=0008 S=B110 D=FF00 B=00 P=34 E=0  00/0687  REP #$10
A=FF04 X=0000 Y=0008 S=B110 D=FF00 B=00 P=24 E=0  00/0689  LDX $B20E (@00B20E F8 F0 08 ...)
A=FF04 X=F0F8 Y=0008 S=B110 D=FF00 B=00 P=A4 E=0  00/068C  TXS
A=FF04 X=F0F8 Y=0008 S=F0F8 D=FF00 B=00 P=A4 E=0  00/068D  LDA $0106 (@000106 00 00 00 ...)
A=FF00 X=F0F8 Y=0008 S=F0F8 D=FF00 B=00 P=26 E=0  00/0690  BNE $06 ($000698)
A=FF00 X=F0F8 Y=0008 S=F0F8 D=FF00 B=00 P=26 E=0  00/0692  LDA $0D,S (@00F105 00 00 00 ...)
A=FF00 X=F0F8 Y=0008 S=F0F8 D=FF00 B=00 P=26 E=0  00/0694  CMP #$00
A=FF00 X=F0F8 Y=0008 S=F0F8 D=FF00 B=00 P=27 E=0  00/0696  BNE $08 ($0006A0)
A=FF00 X=F0F8 Y=0008 S=F0F8 D=FF00 B=00 P=27 E=0  00/0698  PLD
A=FF00 X=F0F8 Y=0008 S=F0FA D=0505 B=00 P=25 E=0  00/0699  PLB
A=FF00 X=F0F8 Y=0008 S=F0FB D=0505 B=00 P=27 E=0  00/069A  REP #$20
A=FF00 X=F0F8 Y=0008 S=F0FB D=0505 B=00 P=07 E=0  00/069C  PLY
A=FF00 X=F0F8 Y=3F1E S=F0FD D=0505 B=00 P=05 E=0  00/069D  PLX
A=FF00 X=0118 Y=3F1E S=F0FF D=0505 B=00 P=05 E=0  00/069E  PLA
A=FBE8 X=0118 Y=3F1E S=F101 D=0505 B=00 P=85 E=0  00/069F  RTI
A=FBE8 X=0118 Y=3F1E S=F105 D=0505 B=00 P=00 E=0  00/0000  STP
A=FBE8 X=0118 Y=3F1E S=F105 D=0505 B=00 P=00 E=0  00/0001  ORA $00,S (@00F105 00 00 00 ...)
A=FBE8 X=0118 Y=3F1E S=F105 D=0505 B=00 P=00 E=0  00/0001  ORA $00,S (@00F105 00 00 00 ...)
A=FBE8 X=0118 Y=3F1E S=F105 D=0505 B=00 P=00 E=0  00/0001  ORA $00,S (@00F105 00 00 00 ...)
A=FBE8 X=0118 Y=3F1E S=F105 D=0505 B=00 P=00 E=0  00/0001  ORA $00,S (@00F105 00 00 00 ...)
--- SNIP ---

Case 2:
Fuzix on (atm latest) commit: af6bd7bd19dcd12206da7bf5070d979c6de8530e
cc65 2.18 release in ubuntu package: cc65 V2.18 - Ubuntu 2.18-1
virtual 65816 from https://github.com/EtchedPixels/

What i did:
Makfile changed target: TARGET=v65c816
Did make and build-mini-filesystem disk0 256 65535 for disk0
Copied disk0 to v65c816 and did make and make os.
Result:
Started v65 and choosing bootdev 0 gives me this:
bootdev: 0
Mounting root fs (root_dev=0,ro): _?  h  Starting /init

panic: no /init

I'm not sure if I'm doing this right or it's a bug?

EtchedPixels, 2020-07-06:
It's a bug. I've not finished getting the 65c816 tree up with the current changes.

EtchedPixels, 2020-07-06:
Loader loads to $FC00 and assumes that'll be enough space. In fact we need more so the loader/kernel packing logic wants changing.

EtchedPixels, 2021-11-09:
Dropping this virtual target

closed 2021-11-09 by EtchedPixels