EtchedPixels/FUZIX #800: no way to transfer files

Jonson26, 2020-11-08:
Currently there is no real way, that I know of, to transfer files from and to a machine running FUZIX. A simple xmodem program included in the os distribution could fix that.

beretta42, 2020-11-11:
i agree.  We could port lrzsz from gnu, or roll our own.

retrac0, 2020-11-12:
Another option would be Kermit.  It's exceptionally portable, and E-Kermit is a pure ANSI C library that should work with a simple wrapper.  It claims 15 KB of code size on x86.

http://www.columbia.edu/kermit/ek.html

jbquirk, 2020-11-18:
Are talking on a running system or as you building the system. If a running system yes I agree some form of serial transfer would be nice but for large files and building a new system I use a loop device under Linux and the tools are built as part of the build process.

KatolaZ, 2021-01-25:
From a running FUZIX system you can use uud/uue and cat directly from/to the serial port (you just need uuencode/uudecode on the linux/BSD side). it's not super-handy, but definitely usable, and does not require another tool. My2Cents.

closed 2021-06-27 by EtchedPixels