EtchedPixels/FUZIX #801: How to build disk images and boot in z80pack?

eshazen, 2020-11-24:
I'm trying to figure out how to boot in z80pack.  I have z80pack 1.36 installed.  It comes with a set of very old images, which don't work.  I'd like to try booting the current repo fuzix.  Question is how to build the disk images.  Issues 168, 169 and 172 are slightly helpful.  172 in particular said (in 2016):

_Known bug - we need a script to build a small floppy of basic tools only. Also ucp blows up when we run out of space.

It's issues 168 and 169

If you put just a kernel on floppy and the image on hard disk and boot with that as root fs it should work, ditto if you hand build a small file system or hack the script to only install useful bits needed for rescue/install not stuff like banner._

Has ucp been fixed?  Could we restore the lost "createdrives" script?  It was removed in 2018 with the comment "createdrives is obsolete".  Otherwise possibly a few words specifying how to build a set of boot disks would be great.

I'll keep struggling and if I figure it out I'll post an answer.

eshazen, 2020-11-25:
OK, much progress.  It boots!  The floppy images are 77 tracks, 26 sectors, 128 bytes each (total size 256256 bytes).  The current bootloader in Kernel/platform-z80pack/bootblock.s expects the OS at track 58, or offset 199680 (0x2f200).  Not sure what else is important on the floppy image, but I did the following:

Started with the fuzix.dsk from the z80pack 1.36 distro.  Patched in the current bootloader and kernel as follows:

$ dd if=fuzix.bin of=fuzix.dsk bs=1 seek=193024 conv=notrunc

$ dd if=bootblock.bin  of=fuzix.dsk bs=1 count=128 conv=notrunc

Then I created a hard drive image by running "build-filesystem-ng" in ...Standalone/filesystem-src.  Watch out, this creates by default a file also called "fuzix.dsk".  Rename to "hd-fuzix.dsk"

Then edit a startup script "fuzix" in the "cpmsim" directory of z80pack so:

#!/bin/sh

rm -f disks/drive[aij].dsk
ln disks/library/eric-fuzix.dsk disks/drivea.dsk
ln disks/library/eric-hd-fuzix.dsk disks/drivei.dsk

./cpmsim $*

It works!  Now it boots tom a login prompt, and I can log in as root.

ojej, 2021-06-16:
any tutorial how run cpu emulator?

closed 2021-06-22 by EtchedPixels