EtchedPixels/FUZIX #802: diff causes a "panic: corrupt inode"

eshazen, 2020-11-29:
I built the Kernel using `sdcc 3.8.0` for `z80pack` with everything set to defaults in the `Makefile`.  Then I created a root filesystem using the script `build-filesystem-ng`.  I created a boot floppy using:

```
dd if=/dev/zero of=$1 bs=1 count=256256
dd if=~/work/FUZIX/Kernel/fuzix.bin of=$1 bs=1 seek=193024 conv=notrunc
dd if=~/work/FUZIX/Kernel/platform-z80pack/bootblock.bin  of=$1 bs=1 count=128 conv=notrunc
```

I start the system with:
```
#!/bin/sh

rm -f disks/drive[aij].dsk
ln disks/library/boot.dsk disks/drivea.dsk
ln disks/library/eric-hd-fuzix.dsk disks/drivei.dsk

./cpmsim $*
```

To trigger the bug:
```
./cpmsim $*
hazen@eric-x20:cpmsim$ ./eric-fuzix

#######  #####    ###            #####    ###   #     #
     #  #     #  #   #          #     #    #    ##   ##
    #   #     # #     #         #          #    # # # #
   #     #####  #     #  #####   #####     #    #  #  #
  #     #     # #     #               #    #    #     #
 #      #     #  #   #          #     #    #    #     #
#######  #####    ###            #####    ###   #     #

Release 1.36, Copyright (C) 1987-2017 by Udo Munk

CPU speed is unlimited

Booting...

-
FUZIX version 0.4pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2020 Alan Cox <alan@etchedpixels.co.uk>
Devboot
484kB total RAM, 420kB available to processes (15 processes max)
Enabling interrupts ... ok.
bootdev: 0
Mounting root fs (root_dev=0, ro): OK
Starting /init
init version 0.9.0ac#1
Checking root file system.

 ^ ^
 n n   Fuzix 0.3.1
 >@<
       Welcome to Fuzix
 m m

login: root

Welcome to FUZIX.
# cat > file1
hello
# cat > file2
hello again
# diff file1 file2
1c1
< hello
---
> hello again

panic: corrupt inode
System halted, bye.
```

EtchedPixels, 2021-06-27:
Doesn't seem to be z80pack specific but diff corrupting memory

EtchedPixels, 2022-12-17:
Obscure kernel bug with unlink()

Fixed

closed 2022-12-17 by EtchedPixels