EtchedPixels/FUZIX #814: qemu and testing

ojej, 2021-01-10:
Is possible starting without small processor?
any qemu script paramaterr for test for people?

For example I have a small procesor tomu.im and how run on it fUZIx

EtchedPixels, 2021-01-10:
For the existing systems the RC2014 emulator and various other emulators for old systems work. I've never seriously looked at ARM and RISC-V and there is no ARM or RISC-V support in the code base at all

Zardoz89, 2021-01-11:
qemu supports 68000 (NeXT and Machintosh Quadra emulation), so could help with 68000 support ?

https://wiki.qemu.org/Documentation/Platforms/m68k

EtchedPixels, 2021-01-11:
The RC2014 emulators already cover tiny68k and related platforms. The Atari ST work though far from finished can also be run on Hatari or other Atari ST emulators. No need for qemu.

ojej, 2021-01-13:
@Zardoz89 Can You show how run fuzix in qemu?

closed 2021-02-17 by EtchedPixels

ojej, 2021-02-28:
hello, can you show how run in emulator?

Zardoz89, 2021-03-01:
I never try it, so I can't

ojej, 2021-04-22:
ok