EtchedPixels/FUZIX #832: uget with ints

davidgiven, 2021-02-23:
So I saw the change to blkdev.c to remove the use of ugetl --- I didn't realise that wasn't available on all platforms. Unfortunately, I've discovered some other occurrences of the same bug, such as in `_pipe()`, which assumes the arguments are `int16_t`s. This was what was causing pipes not to work on both the ESP8266 and the Pico (and presumably on all 32-bit architectures).

However, just adding ugetl won't help because then it'll be the wrong size on 16-bit systems. There really needs to be a ugeti or equivalent, which is either 16 or 32 bits depending on the system.

I've managed to work around this for this specific case but I think there needs to be a more generic solution here.

EtchedPixels, 2021-02-23:
Agreed

davidgiven, 2021-02-24:
Sigh. You can tell I get to do this late at night, can't you? Turns out that ugeti/uputi already exist everywhere, so there is no work that needs to be done here other than to fix pipe(). Closing.

closed 2021-02-24 by davidgiven