EtchedPixels/FUZIX #834: problems with boot fuzix on sc126

exepirit, 2021-02-24:
I tried run FUZIX on SC126 single board computer with RomWBW v3.0 using as bootloader.
FUZIX IDE image flashed on SD card (`dd if=sc126-0.3.ide of=/dev/sdc skip=2`) and putted in computer SD card slot. After power on SBC and select boot from SD card in terminal appears bootloader information and some unreadable symbols:

```

RomWBW HBIOS v3.0, 2020-03-27

SC126 Z8S180-N @ 18.432MHz IO=0xC0
0 MEM W/S, 2 I/O W/S, INT MODE 2
512KB ROM, 512KB RAM

ASCI0: IO=0xC0 ASCI W/BRG MODE=115200,8,N,1
ASCI1: IO=0xC1 ASCI W/BRG MODE=115200,8,N,1
DSRTC: MODE=STD IO=0x0C Sat 2000-01-01 00:00:02 CHARGE=OFF
MD: UNITS=2 ROMDISK=384KB RAMDISK=384KB
IDE: IO=0x10 MODE=RC
IDE0: NO MEDIA
IDE1: NO MEDIA
PPIDE: IO=0x20 PPI NOT PRESENT
SD: MODE=SC OPR=0x0C CNTR=0xCA TRDR=0xCB DEVICES=1
SD0: SDHC NAME=      BLOCKS=0x003B8800 SIZE=1905MB

Unit        Device      Type              Capacity/Mode
----------  ----------  ----------------  --------------------
Char 0      ASCI0:      RS-232            115200,8,N,1
Char 1      ASCI1:      RS-232            115200,8,N,1
Disk 0      MD1:        RAM Disk          384KB,LBA
Disk 1      MD0:        ROM Disk          384KB,LBA
Disk 2      IDE0:       Hard Disk         --
Disk 3      IDE1:       Hard Disk         --
Disk 4      SD0:        SD Card           1905MB,LBA

SC126 Boot Loader

ROM: (M)onitor (C)P/M (Z)-System (F)orth (B)ASIC (T)-BASIC (P)LAY (U)SER ROM  
Disk: (0)MD1 (1)MD0 (2)IDE0 (3)IDE1 (4)SD0 

Boot Selection? 4    Slice(0-9)[0]? 

Booting Disk Unit 4, Slice 0...

Reading disk information...
Loc=F200 End=F400 Ent=F200 Label=Fuzix 126 Loader

Loading...
�������������������������������������
```
Then nothing happens.
Flash root filesystem into partition 1 (`dd if=rootfs-z80-32 of=/dev/sdc1) change nothing. Maybe i flash SD card not correctly or FUZIX does not work properly with SC126?

EtchedPixels, 2021-02-24:
The Fuzix image was designed for ROMWBW 2. ROMWBW 3 changed the default baud rate in the firmware to 115200.

I suspecit it may be working but it's at 38400 baud.

exepirit, 2021-02-25:
Yep, if change baudrate after unknown symbols appears then bootloader print `bootdev:`. Theoritaclly is it possible to rebuild FUZIX with 115200 baud terminal support?

EtchedPixels, 2021-02-26:
You can change the boot baud rate yes

Kernel/platform-*/config.h  TTY_INIT_BAUD

closed 2021-02-28 by exepirit