EtchedPixels/FUZIX #838: Is there a simh (altair z80) port available?

mike632t, 2021-03-07:
I've read the article on [Porting Fuzix To A Banked Z80 Machine](https://github.com/EtchedPixels/FUZIX/blob/master/Documentation/BankedZ80.md) but I have to confess that I'm not really that much wiser about what I'd have to do to get Fuzix running on simh (altair z80) and I'm not sure I want to try just yet!

Unfortunately after several attempts I've not managed to find anything to suggest that this has been done already, so if anyone has any details of an existing simh port let me know.  

Thank you

EtchedPixels, 2021-03-08:
Not aware of anything - but the Altair never had banked memory in most configurations. There is a port to the Cromemco which as most people used similar banking on S100 ought to be similar to what you'd need.

retrac0, 2021-03-08:
It's not the Altair, which out of the box doesn't have the hardware necessary to run FUZIX as Alan notes.  

But SIMH also supports the N8VEM SBC v2, with its bankswitching, which is a FUZIX target.  It might run under that, but I haven't tried it.  I might give it a shot this evening.

https://github.com/simh/simh/blob/9ef15802f0e94ba93af819545565880fe4c9d5c5/AltairZ80/n8vem.c

mike632t, 2021-03-08:
From the documentation I assume that the N8VEM emulation is what you get if you build simh with 'make altairz80' ..?  

In which case that is exactly what I'm after!  Is it really as simple as just booting from a disk image?

drawkula, 2021-03-08:
> From the documentation I assume that the N8VEM emulation is what you get if you build simh with 'make altairz80' ..?

Look inside https://schorn.ch/cpm/zip/n8vem.zip ... `n8vem` is a configuration variant of `altairz80`.

EtchedPixels, 2021-03-09:
My RC2014 emulator can also emulate the SBCv2 including the needed mod for Fuzix to work nicely (the SBCv2 has no interval timer interrupt). That one I know does work with Fuzix in emulation because it is how I did most of the port 8)

drawkula, 2021-03-09:
Is `platform-z80pack` dead?

closed 2021-03-27 by EtchedPixels