EtchedPixels/FUZIX #844: A nasty consequence of side effect in process.c

pawosm-arm, 2021-05-03:
It took me couple of evenings and fair amount of cursing before I've found this:

```
diff --git a/Kernel/process.c b/Kernel/process.c
index 501b8ffa..ad9aa521 100644
--- a/Kernel/process.c
+++ b/Kernel/process.c
@@ -526,6 +526,9 @@ void timer_interrupt(void)
 void unix_syscall(void)
 {
        udata.u_error = 0;
+#ifdef udata
+       susize_t u_retval;
+#endif
 
        /* Fuzix saves the Stack Pointer and arguments in the
         * Assembly Language Function handler in lowlevel.s
@@ -540,7 +543,12 @@ void unix_syscall(void)
                        udata.u_argn1, udata.u_argn2);
 #endif
                // dispatch system call
+#ifdef udata
+               u_retval = (*syscall_dispatch[udata.u_callno]) ();
+               udata.u_retval = u_retval;
+#else
                udata.u_retval = (*syscall_dispatch[udata.u_callno]) ();
+#endif
 
 #ifdef DEBUG_SYSCALL
                kprintf("\t\t\tpid %d: ret syscall %d, ret %p err %p\n",
```
(this `#ifdef` is added here deliberately to prevent any future attempts to optimize this code that could reinstantiate this bug)

This bug has blocked me for good. But finally, I'm observing workable userspace.

EtchedPixels, 2021-05-03:
That surprised me but you are correct: C2011 6.5.16 does permit ass backwards evaluation of those two. Joy

EtchedPixels, 2021-05-04:
Added a temporary variable and comment. The ifdef isn't really needed as the smarter compilers will remove it if it is safe to do so and even the dumb ones seem to mostly figure it out.

pawosm-arm, 2021-05-04:
Are you sure the `rv` is always 8bit?

EtchedPixels, 2021-05-05:
Umm no let's try that again shall we 8)

pawosm-arm, 2021-05-05:
Works for me now. Thanks!

```
Current date is Wed 2021-05-05
Enter new date:
Current time is 21:07:20
Enter new time:

 ^ ^
 n n   Fuzix 0.3.1
 >@<
       Welcome to Fuzix
 m m

login: root

Welcome to FUZIX.
# ls /usr/src/fforth
fforth_tests.fth
# fforth /usr/src/fforth/fforth_tests.fth
TESTING CORE WORDS
TESTING BASIC ASSUMPTIONS
TESTING BOOLEANS: INVERT AND OR XOR
TESTING 2* 2/ LSHIFT RSHIFT
TESTING COMPARISONS: 0= = 0< < > U< MIN MAX
TESTING STACK OPS: 2DROP 2DUP 2OVER 2SWAP ?DUP DEPTH DROP DUP OVER ROT SWAP
TESTING >R R> R@
TESTING ADD/SUBTRACT: + - 1+ 1- ABS NEGATE
TESTING MULTIPLY: S>D * M* UM*
TESTING DIVIDE: FM/MOD SM/REM UM/MOD */ */MOD / /MOD MOD
TESTING HERE , @ ! CELL+ CELLS C, C@ C! CHARS 2@ 2! ALIGN ALIGNED +! ALLOT
TESTING CHAR [CHAR] [ ] BL S"
TESTING ' ['] FIND EXECUTE IMMEDIATE COUNT LITERAL POSTPONE STATE
TESTING IF ELSE THEN BEGIN WHILE REPEAT UNTIL RECURSE
TESTING DO LOOP +LOOP I J UNLOOP LEAVE EXIT
TESTING DEFINING WORDS: : ; CONSTANT VARIABLE CREATE DOES> >BODY
INCORRECT RESULT: { CR1 -> 3 }
INCORRECT RESULT: { ' W1 >BODY -> HERE }
INCORRECT RESULT: { W1 -> HERE 1 + }
INCORRECT RESULT: { W1 -> HERE 2 + }
TESTING EVALUATE
TESTING SOURCE >IN WORD
TESTING <# # #S #> HOLD SIGN BASE >NUMBER HEX DECIMAL
TESTING FILL MOVE
TESTING OUTPUT: . ." CR EMIT SPACE SPACES TYPE U.
YOU SHOULD SEE THE STANDARD GRAPHIC CHARACTERS:
 !"#$%&'()*+,-./0123456789:;<=>?@
ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`
abcdefghijklmnopqrstuvwxyz{|}~
YOU SHOULD SEE 0-9 SEPARATED BY A SPACE:
0 1 2 3 4 5 6 7 8 9
YOU SHOULD SEE 0-9 (WITH NO SPACES):
0123456789
YOU SHOULD SEE A-G SEPARATED BY A SPACE:
A B C D E F G
YOU SHOULD SEE 0-5 SEPARATED BY TWO SPACES:
0  1  2  3  4  5
YOU SHOULD SEE TWO SEPARATE LINES:
LINE 1
LINE 2
YOU SHOULD SEE THE NUMBER RANGES OF SIGNED AND UNSIGNED NUMBERS:
SIGNED: -80000000 7FFFFFFF
UNSIGNED: 0 FFFFFFFF
TESTING INPUT: ACCEPT
TESTING DICTIONARY SEARCH RULES
# free
         total         used         free
Mem:       256           46          210
Swap:        0            0            0
#
```

closed 2021-05-05 by pawosm-arm