EtchedPixels/FUZIX #850: Develop software for FUZIX

tstih, 2021-07-04:
Hello,
I was just wondering if there is any documentation available about developing for Fuzix for Z80 (not developing Fuzix)? Specifically, is there a way to create a relocatable binary for Z80 so that you could run multiple processes at the same time and they would each load on a different address? 
Sincerely,
Tom

EtchedPixels, 2021-07-05:
Fuzix Z80 binaries are passed their load address in a register, which is sufficient for them to then self relocate by any scheme they choose. If you look at Library/libs/crt0_z80_rel.s you'll see an example. Right now that's how all the 8bit platforms handle relocations except the 6502 - which appears to be incapable of self relocating the ZP references so will need a kernel helper.

Note that the kernel doesn't care how you do your relocations so in theory another Z80 toolchain can run with a crt0 that does relocations differently or they can be changed without the kernel caring.

Currently there is no platform that uses the relocations to load multiple binaries into the same bank at once, because the 8bit systems it runs on are banked so you've got multiple banks at the same address anyway. 68K and the like do use this ability when you have a flat memory space.

tstih, 2021-07-06:
Thank you for the explanation, it seems logical to do self relocation for banked system. But there's also an obvious need for a linker that could handle z80 relocatable binaries (in SDCC all relocations are all in the .REL file, the R row, so it does not look like an impossible task). Hence, I have decided to update the linker to include this feature. Will need approx 1 month, and will share the results.

EtchedPixels, 2021-07-07:
I was under the impression that Z88DK's fork of the SDCC toolchain could do this. There are also previous implementations of it that never got into SDCC upstream. At the moment the Fuzix tools do it the traditional way - link the file twice and analyse the resulting binary differences.

SDCC also has other problems in this area
- You can't specify a code/data/etc section starts at 0 (the linker seems to confuse 0 and 'not set')
- it's obsessed with the idea of ROM based programs so you can't tell it to just put initialized data in one place. Again I think the Z88DK version of the tools has that fixed.

https://cpctech.cpc-live.com/download/link.zip

feilipu, 2021-07-07:
Hi @tstih,
What Alan says is quite right. There's quite a lot of SpectrumNext and other banking baked in z88dk. As well as some support for Microsoft PRL. Choice of RAM and ROM builds, with fine grained SECTION assignment. Might also be an option.
Cheers.

tstih, 2021-07-08:
Hello,
I reopened this topic.

Perhaps I don't understand the problem correctly, but you can set code segment location to 0 in SDCC 
sdcc -mz80 --code-loc 0 test.c
The bigger problem is the data segment which you have to set manually. It's possible to do this by double linking. First call to sdcc call generates the .lk file. Then using sed you can remove the line: -b _DATA = 0x8000 from it and just relink passing the modified .lk file. It then creates a code at 0x0000 and places data segment at the end of it. However it still separates the initializer and initialized - leading to two allocations for one variable which is a bit expensive if you start with a lot of initialized data.

Compiling to 0 and 101 and seeking differences will cause problems with SDCC compile, as there are some bit relocations, which this tool can't solve. But they are all there in the .REL file

I followed z88dk debates about relocatable binaries and then they suddenly stopped. This is the last one -
 https://www.z88dk.org/forum/viewtopic.php?p=14604
I asked questions in their z88dk forum (which seems dead), the World of Spectrum and elsewhere and it seems no one knows enough about it for practical use.

The z88dk seems like a good out of the box option, but it lacks compiler documentation. With forums  dead I'm afraid of running out of support and into technical dead ends. Its focus seems to be on libraries and on binaries for known targets, but little is known about the compiler, the various configuration and segmentation options, etc. I'd switch if there was a lively forum like this to ask things and be able to proceed.

feilipu, 2021-07-09:
> I'd switch if there was a lively forum like this to ask things and be able to proceed.

Raise a "Discussion" issue on GitHub. There is enough to do without trolling the forum threads.

Or add to one of the existing issues, then contributors are notified automatically. 

Also attention and capacity varies depending on employment workload I'm afraid.

👍

EtchedPixels, 2021-07-09:
--code-loc 0 works because code defaults to 0.

You don't need to relink to get the data right - just ensure your crt0 mentions the segments you want in the order you need and magic will happen. Hackish but it works.

If you look at the Fuzix tools they know both how to build relocatable binaries by double linking and  how to deal with the initializer/initialized mess providing you've got enough address space to hold both in the initial link.

Another option I've been considering is to teach the assembler/linker kit I used with the 6803/6303 C compiler the rather warped syntax used by SDCC. That would give a rather more sane toolchain and as it captures relocations it wouldn't be hard to teach the linker to output relocatable binaries and SDCC can then just be used for the C to asm phase.

https://github.com/EtchedPixels/CC6303/tree/master/as68

Fuzix always loads Z80 binaries at XX00 because of the fact you may have page aligned tables and other things where the relocation isn't possible - or is invisible (.REL files won't save you from stuff like:

LD HL, TABLE

blah blah

IN L,(C)
LD A,(HL)
LD (DE),.A

type stuff

I believe the CPC stuff is from a much older tool version as well.

closed 2021-07-11 by EtchedPixels