EtchedPixels/FUZIX #851: Another problem compiling for Amstrad NC200

tjewell99, 2021-07-08:
Hi, not under MacOS this time, just Ubuntu. Using SDCC 3.5.0, patched as required:

```
fcc -mz80 -O2 -c tfind.c
tfind.c:25: error 47: indirections to different types assignment   
from type 'void generic* generic* auto'
  to type 'struct node_t generic* generic* auto'
make[1]: *** [Makefile.z80:182: tfind.rel] Error 1
make[1]: Leaving directory '/media/sf_TheArchive/Public/Amstrad/Fuzix/FUZIX/Library/libs'
make[1]: Entering directory '/media/sf_TheArchive/Public/Amstrad/Fuzix/FUZIX/Library/libs'
cp crt0.rel crt0nostdio.rel c.lib /opt/fcc/lib/
cp: cannot stat 'crt0.rel': No such file or directory
cp: cannot stat 'crt0nostdio.rel': No such file or directory
cp: cannot stat 'c.lib': No such file or directory
make[1]: *** [Makefile.z80:212: install] Error 1
make[1]: Leaving directory '/media/sf_TheArchive/Public/Amstrad/Fuzix/FUZIX/Library/libs'
make: *** [Makefile:90: libs] Error 2
```

Again, quite possibly something daft I've done!

EtchedPixels, 2021-07-09:
Ok where does it still mention 3.5 as I need to fix that. You need at least 3.8/3.9 or so at this point.

tjewell99, 2021-07-09:
Thanks! It's here - Kernel/patches/SDCC - you mention 3.5.0

I did wonder if I needed a later version, but when I tried to patch 3.8.0, the patch for sdcc/src/SDCCglue.c failed, so I presumed I should use 3.5.0. 

I've patched 3.8.0 again, ignoring the failed chunk this time, and I get much further. The libraries compile, but I have issues later.

```
make[1]: Leaving directory '/media/sf_TheArchive/Public/Amstrad/Fuzix/FUZIX/Library/libs'
(cd Applications; make)
make[1]: Entering directory '/media/sf_TheArchive/Public/Amstrad/Fuzix/FUZIX/Applications'
(cd util; make -f Makefile.z80)
make[2]: Entering directory '/media/sf_TheArchive/Public/Amstrad/Fuzix/FUZIX/Applications/util'
../../Library/tools/fcc -mz80  -O2 -c banner.c
../../Library/tools/fcc -mz80   banner.rel -o banner

?ASlink-Warning-Undefined Global '__divuint' referenced by module 'fread'

?ASlink-Warning-Undefined Global '__mulint' referenced by module 'fread'

?ASlink-Warning-Undefined Global '__modulong' referenced by module 'ltostr'

?ASlink-Warning-Undefined Global '__divulong' referenced by module 'ltostr'

?ASlink-Warning-Undefined Global '___sdcc_call_hl' referenced by module 'atexit'

?ASlink-Warning-Undefined Global '_memcpy' referenced by module 'fread'

?ASlink-Warning-Undefined Global '_strcat' referenced by module 'vfprintf'

?ASlink-Warning-Undefined Global '_strlen' referenced by module 'banner'

?ASlink-Warning-Undefined Global '_strlen' referenced by module 'vfprintf'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'banner'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'gets'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'printf'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'fputc'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'atexit'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'vfprintf'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'fputs'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'fflush'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'fgetc'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'lseek'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'ltostr'

?ASlink-Warning-Undefined Global '___sdcc_enter_ix' referenced by module 'fread'

?ASlink-Warning-Undefined Global '__modsint' referenced by module 'banner'
make[2]: *** [Makefile.z80:154: banner] Error 1
make[2]: Leaving directory '/media/sf_TheArchive/Public/Amstrad/Fuzix/FUZIX/Applications/util'
make[1]: *** [Makefile:10: util] Error 2
make[1]: Leaving directory '/media/sf_TheArchive/Public/Amstrad/Fuzix/FUZIX/Applications'
make: *** [Makefile:94: apps] Error 2
```

EtchedPixels, 2021-07-09:
File updated to mark it obsolete
You should be able to use a standard modern SDCC for all but banked kernels, and for those there is a github tree.

The errors there looks like you've got 3.5 and 3.8 code muddled together.

tjewell99, 2021-07-09:
Ah, just spotted issue #158 - this could be because my paths are in a bit of a mess with the different versions of SDCC. Let me clean up and see if I can make this behave.

tjewell99, 2021-07-09:
All compiled okay! Very nearly got it running - a possible other problem, but unrelated to this. I'll start a new issue if it's a real problem. Thanks for your help!

closed 2021-07-09 by tjewell99