EtchedPixels/FUZIX #853: SC102 - Warning: no timer available with 20Mhz Z80.

Spoozilla, 2021-07-11:
I have a bit of a mix and match RC2014 setup which includes an SC102, a #22c z80 CPU module from Rotten Snow, and a #33 ACIA serial board amongst others. I'm using the SC102 oscillator for the bus and the ACIA, which works as expected. As explained in the source I have chained the CTC output from CH2 to the input of CH3. If I run the CPU from the bus clock then the CTC is detected and seems to work as expect. However, if I switch the z80 to using it's own 20Mhz clock (with the rest of the bus still using the 7Mhz clock) then the CTC is not detected.

Also, unrelated, but it's worth noting that RomWBW now seems to default the narrow ACIA to 0x80, not 0xA0 as FUZIX expects.

EtchedPixels, 2021-07-11:
That I would expect - the Z80 CTC/PIO/SIO/DMA only work properly when they are running on the CPU clock. It's a design limitation of the devices.

I'll double check the ACIA. I went with what Wayne said when he did the changes but I've not checked what actually happened as I used 16x50's for pretty much everything.

Spoozilla, 2021-07-11:
Ok, thanks for clearing up my misunderstanding. I thought they depended on the 7.37 Mhz bus signal. I guess the documentation is assuming there being only a single clock in the system. It's not a problem, it's working fine... the extra 13 Mhz would have been nice though.

EtchedPixels, 2021-07-13:
So I did some testing and then talked to Wayne about ROMWBW

At the moment 2.9.x expects A0
3.0.x is simply broken with a narrow ACIA

Once there is an actual ROMWBW fix I'll follow it up further

Spoozilla, 2021-07-13:
I'm sorry to be contrarian but that's simply not my experience at all, especially given that this machine is running RomWBW 3.0.1. If I address the ACIA to 0xA0-0xA7 I get no serial connection, but at 0x80-0x87 the machine boots as expected. I then patched the ACIA_C and ACIA_D addresses inrc2014.s to match this. Since I have the CTC in the default port of 0x88 this would indicate that the addressing space is narrow.

Running survey from within CP/M backs this up.

```
B>survey                                                                        
                *** System Survey (December 17) ***                             
                                                                                
Drive A: 24K bytes in 0 files with 360K bytes remaining                         
Drive B: 378K bytes in 57 files with 6K bytes remaining                         
                                                                                
Memory map:                                                                     
0       8       16      24      32      40      48      56      64              
|       |       |       |       |       |       |       |       |               
 TTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTTCCCBBBBBBB               
T=TPA   C=CPM   B=BIOS or unassigned    R=ROM or bad                            
BIOS at E603    iobyte 94       drive 01        BDOS at D806                    
                                                                                
65535 Bytes RAM         0 Bytes ROM             55302 Bytes in TPA              
0 Bytes Empty           65535 Total Active Bytes                                
                                                                                
Active I/O ports:                                                               
17                                                                              
80 81 82 83 84 85 86 87 88 89 8A 8B                                             
13 Ports active 
```

Here's the boot output including a Fuzix boot.

```
RomWBW HBIOS v3.0.1, 2020-06-30                                              
                                                                             
RC2014 Z80 @ 7.372MHz                                                        
0 MEM W/S, 1 I/O W/S, INT MODE 1        
512KB ROM, 512KB RAM                    
                                        
ACIA0: IO=0x80 ACIA MODE=115200,8,N,1   
DSRTC: MODE=STD IO=0xC0 Tue 2021-07-13 13:51:52 CHARGE=ON
MD: UNITS=2 ROMDISK=384KB RAMDISK=384KB 
IDE: IO=0x10 MODE=RC
IDE0: 8-BIT LBA BLOCKS=0x0001E900 SIZE=61MB
IDE1: NO MEDIA
PPIDE: IO=0x20 PPI NOT PRESENT                                                  
                                                                                
Unit        Device      Type              Capacity/Mode                         
----------  ----------  ----------------  --------------------                  
Char 0      ACIA0:      RS-232            115200,8,N,1                          
Disk 0      MD1:        RAM Disk          384KB,LBA                             
Disk 1      MD0:        ROM Disk          384KB,LBA                             
Disk 2      IDE0:       CompactFlash      61MB,LBA                              
Disk 3      IDE1:       Hard Disk         --                          
RC2014 Boot Loader                                                              
                                                                                
ROM: (M)onitor (C)P/M (Z)-System (F)orth (B)ASIC (T)-BASIC (P)LAY (U)SER ROM    
Disk: (0)MD1 (1)MD0 (2)IDE0 (3)IDE1                                             
                                                                                
Boot Selection? 2    Slice(0-9)[0]? 0                                           
                                                                                
Booting Disk Unit 2, Slice 0...                                                 
                                                                                
Reading disk information...                                                     
Loc=F200 End=F400 Ent=F200 Label=Fuzix WBW Loader                               
                                                                                
Loading...FUZIX version 0.4pre1                                                 
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza             
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha         
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>                   
Copyright (c) 2014-2021 Alan Cox <alan@etchedpixels.co.uk>                      
Devboot                                                                         
Z80 CTC detected at 0x88.                                                       
DS1302 detected at 0xC0.                                                        
tty1: ACIA at 0x0080.                                                           
512kB total RAM, 432kB available to processes (20 processes max)                
Enabling interrupts ... ok.                                                     
IDE drive 0: hda: hda1 hda2 (swap)                                              
IDE drive 1: ide error, status=0000                                             
SD drive 0: no card found                                                       
[... cut ...]
                                                                                
 ^ ^                                                                            
 n n   Fuzix 0.3.1                                                           
 >@<                                                                            
       Welcome to Fuzix                                                         
 m m                                                                            
                                                                                
login:
```

EtchedPixels, 2021-07-13:
I'm using the provided ROM images and I've got traces and Wayne's confirmation - it might work on your box but it's buggy. Easy enough to fix once there's a definitive answer and I can test it - as you've already done.

EtchedPixels, 2021-07-14:
(keeping open so I remember to make the change)

EtchedPixels, 2022-02-03:
ROMWBW development is using 0x80 for ACIA and it works correctly that way nowdays. Kernel updated accordingly

closed 2022-02-03 by EtchedPixels