EtchedPixels/FUZIX #855: Fpu.h is missing.

kazueda, 2021-07-24:
Dear SIr or Madam,
When I tried to build new fuzix kernel from the recent source files, the compiler said there is no fpu.h.
Sincerely yours,

EtchedPixels, 2021-07-25:
Thank you for the bug report - I have pushed the file now.

kazueda, 2021-07-25:
Thank you for your fpu.h.
I compiled your updated source file.
The compile stopped with following message.

make[1]: *** [Makefile:40: net_w5x00.rel] Error 1
make[1]: Leaving directory '/home/kazueda/FUZIX/Kernel/platform-sc126'
make: *** [Makefile:114: target] Error 2

Sincerely yours,
kazueda

EtchedPixels, 2021-07-26:
Not sure why - but I pushed all my updates at this point. If it still fails can you include more of the messages before the "Error 1"

Thanks

mennozon, 2021-07-27:
I ran to this as well while building for sc126 (curious to test w5500), I think this is the complete error.

> sdcc -c --std-sdcc99 --no-std-crt0 -mz180 -I/home/dev/source/FUZIX/Kernel/cpu-z180 -I/home/dev/source/FUZIX/Kernel/platform-sc126 -I/home/dev/source/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /home/dev/source/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ -I../dev/net/ -c ../dev/net/net_w5x00.c
discard.c:74: warning 112: function 'netdev_init' implicit declaration
discard.c:74: warning 84: 'auto' variable 'netdev_init' may be used before initialization
discard.c:74: warning 84: 'auto' variable 'netdev_init' may be used before initialization
../dev/devsd_discard.c:110: warning 158: overflow in implicit constant conversion
../dev/devsd_discard.c:118: warning 158: overflow in implicit constant conversion
../dev/devsd_discard.c:121: warning 158: overflow in implicit constant conversion
../dev/net/net_w5x00.c:756: warning 85: in function netproto_setup unreferenced function argument : 's'
../dev/net/net_w5x00.c:816: warning 85: in function netproto_accept_complete unreferenced function argument : 's'
../dev/net/net_w5x00.c:929: error 98: conflict with previous declaration of 'netproto_write' for attribute 'type' at /home/dev/source/FUZIX/Kernel/include/netdev.h:203
from type 'int function ( struct socket generic* fixed, struct ksockaddr generic* fixed) fixed'
  to type 'int function ( struct socket generic* fixed, struct ksockaddr generic* fixed) fixed'
../dev/net/net_w5x00.c:972: error 98: conflict with previous declaration of 'netproto_shutdown' for attribute 'type' at /home/dev/source/FUZIX/Kernel/include/netdev.h:205
from type 'int function ( struct socket generic* fixed, unsigned-char fixed) fixed'
  to type 'int function ( struct socket generic* fixed, unsigned-char fixed) fixed'
../dev/net/net_w5x00.c:998: error 98: conflict with previous declaration of 'netproto_ioctl' for attribute 'type' at /home/dev/source/FUZIX/Kernel/include/netdev.h:209
from type 'int function ( struct socket generic* fixed, int fixed, unsigned-char generic* fixed) fixed'
  to type 'int function ( struct socket generic* fixed, int fixed, unsigned-char generic* fixed) fixed'
make[2]: *** [Makefile:40: net_w5x00.rel] Error 1
make[2]: Leaving directory '/home/dev/source/FUZIX/Kernel/platform-sc126'
make[1]: *** [Makefile:114: target] Error 2
make[1]: Leaving directory '/home/dev/source/FUZIX/Kernel'
make: *** [Makefile:102: kernel] Error 2
make: *** Waiting for unfinished jobs....

EtchedPixels, 2021-07-28:
Curious - my compiler doesn't error them, but I can see why it errors I think. Pushed another change

What version of sdcc are you using ?

mennozon, 2021-07-28:
Using sdcc 4.0.0 on a Debian bullseye machine.

I can confirm it builds now with your latest changes, thanks!

closed 2021-07-29 by EtchedPixels