EtchedPixels/FUZIX #862: build fails

NollKollTroll, 2021-08-18:
Environment:
Linux Mint 20.1
cc65 2.19 from 2 days ago
Fuzix main from <1h ago

Changed the makefile to build for rc2014-6502

Get the following error at the end:

make[1]: Leaving directory '/home/adam/work/FUZIX/Library/libs'
make[1]: Entering directory '/home/adam/work/FUZIX/Library/libs'
make[1]: Nothing to be done for 'install'.
make[1]: Leaving directory '/home/adam/work/FUZIX/Library/libs'
(cd Applications; make)
make[1]: Entering directory '/home/adam/work/FUZIX/Applications'
(cd util; make -f Makefile.6502)
make[2]: Entering directory '/home/adam/work/FUZIX/Applications/util'
cl65 -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502   -c -o cpuinfo.o cpuinfo.c
ca65   -o cpuinfo-6502.o cpuinfo-6502.s
/home/adam/work/FUZIX/Library/link/ld6502 -o cpuinfo -L/home/adam/work/FUZIX/Library/libs /home/adam/work/FUZIX/Library/libs/crt0_6502.o cpuinfo.o cpuinfo-6502.o c6502.lib
/home/adam/work/FUZIX/Library/link/ld6502: 9: [[: not found
cl65: No input files
cl65: No input files
a.out.b1: No such file or directory
make[2]: *** [Makefile.6502:135: cpuinfo] Error 1
make[2]: Leaving directory '/home/adam/work/FUZIX/Applications/util'
make[1]: *** [Makefile:10: util] Error 2
make[1]: Leaving directory '/home/adam/work/FUZIX/Applications'
make: *** [Makefile:99: apps] Error 2

EtchedPixels, 2021-08-18:
I am guessing that Mint doesn't have bash as the default shell ?
If you edit the ld6502.c file and replace the [[ ]] with [ ] it should all still work fine - does that help ?

NollKollTroll, 2021-08-18:
I just tested in the terminal:
ps -p $$
    PID TTY          TIME CMD
 105312 pts/0    00:00:00 bash
So it seems I am running bash. The change to [] helped me get farther, again I did _make clean_ before _make all_. Still stuck a bit later.
BUT when I had a closer look at ld6502, it said #!/bin/sh which when changed to #!/bin/bash ran MUCH farther when using the original [[]]. But still fails.

make[2]: Entering directory '/home/adam/work/FUZIX/Applications/V7/cmd/sh'
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  args.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  blok.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  builtin.c
builtin.c:8: Warning: Parameter 'argn' is never used
builtin.c:8: Warning: Parameter 'cmd' is never used
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  cmd.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  ctype.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  error.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  expand.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  fault.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  io.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  macro.c
macro.c:33: Warning: Control reaches end of non-void function [-Wreturn-type]
macro.c:58: Warning: Control reaches end of non-void function [-Wreturn-type]
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  main.c
main.c:169: Warning: Control reaches end of non-void function [-Wreturn-type]
main.c:228: Warning: Result of comparison is always true
main.c:250: Warning: Result of comparison is always true
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  msg.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  name.c
name.c:298: Warning: Parameter 'n' is never used
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  print.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  service.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  setbrk.c
setbrk.c:18: Warning: Distinct pointer types comparing 'unsigned char *' with 'void *'
setbrk.c:22: Warning: Pointer conversion to 'char *' from 'unsigned char *' changes pointee signedness
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  stak.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  string.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  word.c
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502  xec.c
cl65: Subprocess 'cc65' aborted by signal 11
make[2]: *** [Makefile.6502:26: xec.o] Error 1
make[2]: Leaving directory '/home/adam/work/FUZIX/Applications/V7/cmd/sh'
make[1]: *** [Makefile:16: sh] Error 2
make[1]: Leaving directory '/home/adam/work/FUZIX/Applications'
make: *** [Makefile:99: apps] Error 2

NollKollTroll, 2021-08-18:
Trying to decipher this, the file xec.c seems to be missing from FUZIX/Applications/V7/cmd?

EtchedPixels, 2021-08-18:
cl65 -c -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502 xec.c
cl65: Subprocess 'cc65' aborted by signal 11

Your compiler crashed. If that's repeatable you've hit a bug in the cc65 development tree head. Might be worth trying an actual release ?

mrdudz, 2021-08-18:
the cc65 git head is absolutely required (and it worked for me a couple weeks ago). the earlier release versions will NOT work.

EtchedPixels, 2021-08-18:
The release versions in Fedora have been working fine for me for several releases.

mrdudz, 2021-08-18:
Whatever ancient ones they have - the 2.19 release didnt work for me, git head did. I highly recommend using it too, there were plenty fixes recently.

NollKollTroll, 2021-08-18:
I'll try other versions tomorrow. It is repeatable on the version I have from head.

EtchedPixels, 2021-08-18:
Ah makes sense - Fedora still ships 2.18, maybe they had a good reason to stay there 8)

mrdudz, 2021-08-18:
i seriously doubt it, just coincidence. As said, git head generally is far superior, tons of bugs have been fixed since then.

@NollKollTroll it would be nice if you'd open tickets at the cc65 repo for the files that make cc65 crash so someone can look at it

NollKollTroll, 2021-08-19:
I did some more testing this morning.
A fresh Fuzix pull, and cc65 pull of V2.18 / V2.19 / master. Changed Fuzix target to rc2014-6502.
So far I get the same crash from master, but V2.18 & V2.19 gives another error:
cl65 -t none -O -D__STDC__ -c -O -I/home/adam/work/FUZIX/Library/include -I/home/adam/work/FUZIX/Library/include/6502   -c -o find.o find.c
find.c(1213): Error: Too many local variables
find.c(1213): Error: Too many local variables
find.c(1219): Error: Too many local variables
find.c(1220): Error: Too many local variables
find.c(1232): Error: Too many local variables
find.c(1232): Error: Too many local variables
find.c(1240): Error: Too many local variables
find.c(1240): Error: Too many local variables
find.c(1242): Error: Too many local variables
find.c(1242): Error: Too many local variables
find.c(1245): Error: Too many local variables
find.c(1245): Fatal: Too many errors

Will continue testing.

mrdudz, 2021-08-19:
You'll have to skip a few things in the tree (i did that too) - cc65 has some limitations, eg it can only use 256 bytes maximum for local variables,

in my older local repo i have 4 modified files, with those i could build an image (but never managed to actually run it due to missing instructions...)

[fuzix.zip](https://github.com/EtchedPixels/FUZIX/files/7013677/fuzix.zip)

NollKollTroll, 2021-08-19:
Thanks mrdudz, will try the files tomorrow. My day ended earlier than I hoped...

EtchedPixels, 2021-08-19:
Can't duplicate that one either. What does mint default to for yacc - byacc or is it trying to pretend GNU Bison is yacc ?

NollKollTroll, 2021-08-20:
I installed byacc as that was a prerequisite according to the Fuzix docs. Now when I run yacc I get Bison yacc 3.5.1, byacc gives the proper byacc 1.9. Don't know if that matters or how to choose which one to use.

EtchedPixels, 2021-08-20:
It seems all the Linux distros install Berkeley yacc is byacc, and bison as bison then link one or the other, debian even seems to be you choose. I'll update the rules file to use "byacc" and see if that does the trick, or if it breaks someone else.

NollKollTroll, 2021-08-20:
More testing done now. The files from mrdudz were injected into Fuzix tree.
Build with latest cc65 crashes at the same file, xec.c, I'll file an issue to cc65 about this.
Build with cc65 V2.19 finishes correctly.

EtchedPixels, 2021-08-20:
Pushed a change to Makefile so we default to "byacc"

NollKollTroll, 2021-08-20:
Did a fresh git clone, changed platform to rc2014-6502, then make all. Success!
I believe it is time to close this issue, thanks for all the help.
NOTE: it is only possible to build with cc65 V2.19, master still crashes trying to cl65 xec.c

EtchedPixels, 2021-08-20:
thanks for the update

closed 2021-08-20 by EtchedPixels