EtchedPixels/FUZIX #865: Apple //e Port

retrac0, 2021-08-20:
After nibbling away at the PDP-11 and Macintosh 512K as targets without much luck, I changed tack.  Toolchain issues with the -11 and the Mac deserves FUZIX as an app that runs in a window.  I lost interest in the bare metal approach.  The early boot environment is also very poorly documented.  Though I did write a bootloader.  Will publish if interest.

Anyway, I got some more bite with the //e.  Just some thoughts on the port.

I've written a bootloader - two stage for 140K disks which loads a 4K program, currently in C.  The idea would be to also write 1st stage loaders for the various supported boot devices (Smartport, etc.)   That program can read more Disk II sectors.  I propose bypassing ProDOS and booting directly.  We can then read in the parts where they need to go.  I am currently working on relocating the loader into the language card so it's as out of the way as possible.  That area can be reused as disk buffers or similar after, maybe.

Disk II routines are relatively simple for read, at least slowly one sector at a time.  There seem to be a variety of more sophisticated routines out there, perhaps some under open licenses.  The Apple II can read up to an entire track written as a single long block, in one go.  A ~48 KB of process could be swapped to a Disk II in about half a second, theoretically.  Less if the drive was already spinning.

There is some slack in the Disk II routines with regard for timing for polled IO.  It might be possible with fancy timing to turn interrupts off and still not drop characters with a serial port.  I'm not sure there's anyway to prevent data corruption if you affect read timing with interrupts on.  There is no strong error correction in that regard.  It's improbable but would probably lead to devilish stray errors.  Smartport devices don't need to worry about that.

The proposed memory map makes sense to me.  I'm currently targeting an unenhanced //e with RAMWorks and 2x Disk II drives for /, usr, var, swap.  Not beyond loading an entirely broken kernel yet.

FUZIX has changed a lot since the sketch //e code was written.  I know the 6502 code gets less attention.  Anything I should beware of as I move forth?

EtchedPixels, 2021-08-20:
There are a a bunch of changes - in particular the newproc->makeproc change, tty behaviour changes, relocatable executables and the fact the asm UDATA macros are now offsets not absolute. I've pushed the obvious changes I can see.

EtchedPixels, 2021-08-20:
If the loader hides in the secondary language area then it might make sense rather than killing it off to just let it provide the low level disk routines it needs to have anyway and call then from the OS side ? Would just need to make sure nothing for disk I/O ends up in the same address range - which ideally would be the case.

Serial and floppy disk interrupts is just one of those things. A lot of the 8080 and Z80 machines may have floppy disk controllers but almost none of them have DMA so they still only have 64us or whatever to respond to a disk byte appearing.

EtchedPixels, 2021-08-20:
Looking in more detail at the memory mapping and the fact cc65 can't really generate banked binaries I'm even more of the view that it might make a lot more sense to have a "BIOS" that lived in the alt top bank and provided
- Serial interface glue
- Disk I/O (all it needs is read/write a number of 512 byte blocks  at a given LBA from/to the address given below C000) - be that 140K disks, smartport, the Apple II I/O style ramdisk (which looks ideal for swap) etc
- Possibly the video emulation
- Mouse?
- Timer

That would allow it to all be tested separately and keep the memory map somewhat saner.

retrac0, 2021-08-20:
Alt top bank being the second bank at D000 - DFFF right?  That's what I'm using.  It's 4 KB and Disk II and smartport and serial glue should definitely fit.  Not sure about the rest.  Haven't looked at the mouse card except as the timer.  Encapsulating all the Apple II hardware makes sense but could be a squeeze for the 4 KB bank.  

Standard Apple II ROM routines can be used -- the D000 - DFFF is not used by the Monitor routines up in Fxxx (I believe).  So the built-in routines for console display are what I'm using for now.  

This BIOS shim would need an API for the kernel to call.  Could use a jump table.  Or maybe link them together as part of the kernel then generate as separate binaries?

retrac0, 2021-08-20:
Re: Apple non-RAMWorks expansion (Slinky) cards -- definitely ideal for swap.  They have a post-increment mode that's faster than a block copy would be.

EtchedPixels, 2021-08-20:
I think ld65 can generate two sets of binary blobs for the same address. I did play with it a bit when looking at teach cc65 to generate bankable binaries but didn't play with the feature much.

mrdudz, 2021-08-20:
Yes, it can totally do this. It also has some support for referring to banks in code now, so you can write a "far call" stub and use that to call code in banked memory.

retrac0, 2021-08-22:
I was wrong about using ROM routines.  The LC can only switch in the RAM or ROM address space at a time.  I've started writing my own 40 column console driver.  The Disk II code should weigh in about 1 KB code.  I'm going to stop there and then try to get the kernel at least printing and loaded in a sensible place.  I am going to put Ramworks on hold.  For now 1 process in RAM, target the Apple Memory Exp. as the first block device besides Disk II read, for swap and maybe RAM disk R/W.

retrac0, 2021-08-23:
The Slinky (A II Memory Exp.) is looking increasingly attractive.   Up to 1 MB fast page access.  I'm making it a priority as a block device.  Disk II code should come in around 1 KB.

EtchedPixels, 2022-02-02:
Closing idle

closed 2022-02-02 by EtchedPixels