EtchedPixels/FUZIX #877: Unable to pass bootdev: prompt

anjennings, 2021-09-30:
I am attempting to port Fuzix to a custom SBC and encountering difficulty passing the bootdev prompt. FUZIX.BIN is booted via CP/M as a com file and takes over shortly after boot.

a>d:fuzix

FUZIX version 0.4pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2021 Alan Cox <alan@etchedpixels.co.uk>
Devboot
400kB total RAM, 352kB available to processes (7 processes max)
Enabling interrupts ... ok.
bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: Mounting root fs (root_dev=6, ro): failed
bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: bootdev: 1

Needless to say I'm puzzled by what seems to be happening here.  No CONFIG_IDE/FD/PPIDE is set in config.h as all disk IO is handled through its own non-standard interface involving a propeller and SD card.  This "device" and its functions are in index 1 of the dev_tab. CMDLINE is NULL, so I am not sure why or where root_dev is being set to 6, although it is always 6 when I retry.  The final bootdev allows a user to input a string but pressing return does nothing.  The 16C550 is not interrupt driven and polled during idle and interrupts (although neither seem to be called until after the several dozen bootdevs) and init is minimal as nothing changes between the switch from CP/M to Fuzix.  

I can answer more questions if need be but can anyone offer me any clues as to what is causing this? Thanks

EtchedPixels, 2021-10-01:
That looks like the tty code is handing back lots of junk which happens to include something that has 6\n in it. If it's not calling the serial driver at that point then presumably the tty queue is not set up correctly, corrupted or not properly initialized. That might point to the data segment not being set up properly or the BSS not being initialized as a first guess.

anjennings, 2021-10-01:
I cleaned up some of my driver code and rebuilt the kernel and that did the trick.  
If I may keep this issue open, I am still debugging the init process further down the line:

```
FUZIX version 0.4pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2021 Alan Cox <alan@etchedpixels.co.uk>
Devboot
FUZIX for Tom80, 2021 Aidan Jennings
400kB total RAM, 352kB available to processes (7 processes max)
Enabling interrupts ... ok.
bootdev: 300
Mounting root fs (root_dev=300, ro): fp->s_mounted=0x31C6, fp->s_isize=0x0100, fp->s_fsize=0xFFFF
OK
Starting /init
start execve
staring execution (doesn't return)
init version 0.9.0ac#1
init: passed arg chk
init: passed sbrk
init: passed l_inittab
init: passed p_inittab
init: do runlvl 1
init: do runlvl 2
init: fork...
init: child of fork
init: exe sh
start execve
DBG: HEADER MAGIC, have 28021, want -32600
DBG: header not ok
nogood2
nogood
init: ok try exe sh again
start execve
staring execution (doesn't return)
start execve
DBG: HEADER MAGIC, have 28021, want -32600
DBG: header not ok
nogood2
nogood
start execve
DBG: ino is 0
start execve
staring execution (doesn't return)
start execve
staring execution (doesn't return)
Cannot open file
```

The two attempts in init.c to spawn a shell process seem to fail because of a mismatched header magic number.  I have generated my own disk image and used the 0.4pre image supplied on the website and both have given the same results.  Am I barking up the wrong tree with this?  I'm confused as to why the magic number would be mismatched

EtchedPixels, 2021-10-02:
So _execve() clearly works because your init process loaded. Syscalls are working as it printed the first stuff.

What happens if you run the /bin/sh binary as init ?

anjennings, 2021-10-02:
If I understand you correctly, I changed the call to add_argument in start.c to
```
add_argument("/bin/sh");
```
where it was originally "/init"

This causes a "panic killed init" after the call to execv

I also tried leaving start.c as is and replaced the init binary with sh and rebuilding the fs; that produced the same result

EtchedPixels, 2021-10-03:
If you replace /bin/sh with /bin/ssh then start with the "s" option on the boot line do you get an ssh command line and if so does it let you do anything ?

Trying to work out if its fork or memory map related - what is the memory map like ?

closed 2021-11-08 by EtchedPixels