EtchedPixels/FUZIX #878: RPi Pico build error (square bracket invalid in preprocessor expressions)

Arawn-Davies, 2021-10-01:
Hello, I'm new to FUZIX and I'm following the RPi Pico build documentation, but when I run ```make image -j``` it gives me the error:
```
/home/arawn/fuzix/Kernel/platform-rpipico/./config.h:46:37: error: token "[" is not valid in preprocessor expressions
   46 | #define PROGBASE ((uaddr_t)&progbase[0])
      |                                     ^
```

I've installed the pico sdk and arm-none-eabi toolchain (version 9.2.1) as it's the version that is in Ubuntu WSL2s apt repos, but I don't know GCC processor expressions enough to know if the square brackets were deprecated in a newer version than when the guide was written. 

I know the Pi Pico may not be the best of platforms that FUZIX targets, but this issue is stopping me from at least trying it out, no prebuilt binaries available

@davidgiven thanks for creating the fork! cheaper alternative than buying a whole Z80 SBC just for FUZIX 👌

Arawn-Davies, 2021-10-01:
should note that I did update the path to the pico sdk in the Makefile...

EtchedPixels, 2021-10-01:
You still need to build from @davidgiven 's branch for Pico.

davidgiven, 2021-10-02:
I believe the pico changes should all be upstreamed these days.

AFAICT, somebody somewhere is doing `#if PROGBASE` instead of `#ifdef PROGBASE`. This will be causing the symbol to be interpreted as a preprocessor expression, which isn't valid. This sounds like bitrot since the fork got merged.

EtchedPixels, 2021-10-02:
There are a couple backed out because they broke other builds

Arawn-Davies, 2021-10-02:
> You still need to build from @davidgiven 's branch for Pico.
fork has been merged to upstream, tried building the original repo and I'm only getting fuzix.uf2 not filesystem.uf2, flashed fuzix.uf2 anyway but on USB device appears in Device Manager (I'm using Ubuntu in WSLv2 with compiler installed from apt-get) 

if bitrot is an issue, I'll fiddle about with source to be similar to other/newer platforms but mileage may be low as uni starts Monday and time will be at a premium :/

davidgiven, 2021-10-29:
Sorry, been on holiday. I'll have a look at this, but my Pico SDK has stopped working for some description.

Also, I see that somebody's set up some CI --- I'll try and get that working for the Pico.

closed 2022-02-03 by EtchedPixels