EtchedPixels/FUZIX #898: RC2014-6502 Diskimage fail

floobydust, 2022-01-14:
I'm trying to build a diskimage for the RC2014-6502 profile. I've created a new VM using Ubuntu 21.10 and installed everything listed in the install and build requirements. This includes GCC, Make, Byacc, CC65 and used Git to clone the FUZIX code.

"make" by itself or "make all" completes without error and appears to build a kernel at 64512 bytes. There are numerous warnings however, but no errors. When I use "make diskimage", it fails with 9 errors, all showing missing files. Here's what shows up in the final compilation when it fails:

_Processing package MWC-cmd
ERROR du:10 file not found
Processing package util-mini
Processing package util
ERROR decomp16:70 file not found
ERROR gptparse:78 file not found
ERROR sed:96 file not found
ERROR vile:117 file not found
ERROR ../../Applications/util/fuzix-util.pkg:118 file not found /bin/vile
Processing package adventure
Processing package flashrom
Processing package games
ERROR fortune.dat:23 file not found
ERROR taylormade:43 file not found
Processing package SmallC
Processing package V7-games
Processing package V7-games-man
Processing package V7-cmd-micro
Processing package V7-cmd
ERROR dc:17 file not found
Processing package V7-cmd-man
Processing package V7-sh
Processing package V7-fsh
Processing package V7-sh-man
ERROR Fix errors and re-run
make: *** [Makefile:115: diskimage] Error 2_

I'm not a C guru programmer by any means, but I can't seem to figure what I'm doing wrong or perhaps what I'm missing in either the development environment or the code base. Can someone show me the error of my ways?

Regards, KM

EtchedPixels, 2022-01-17:
Will try and replicate it when I get some time. Some things are not getting built so there ought to be compiler errors in the log (or maybe a segmentation fault or similar if the compiler in ubuntu broke)
fortune.dat missing is odd because that's host generated so may be a different problem to the others

floobydust, 2022-01-18:
Thanks for your reply. I've tried running 3 different versions of Ubuntu (18.04, 20.04 LTS and current) and a current Fedora release. I've even compiled and installed CC65 on one build as well as installing as a package. Oddly, I get identical results in every VM/build. So I'm not sure if it's something I'm missing or doing wrong. I was hoping that some other folks might be trying the same.

I'm in the process of building a serial/parallel/timer card for a RC2014 bus using a 16C550 and 65C22 (single card) and a 65C02 CPU board as well. I currently have a RC2014 ZED running Fuzix with the ROMWbW, so the next step is to try a 6502 build, using the same platform and swapping out some cards.

 Regards, KM

EtchedPixels, 2022-02-02:
Should be fixed now.

Tested with cc65 V2.18 - Fedora 2.18-14.fc34

closed 2022-02-03 by EtchedPixels

floobydust, 2022-02-04:
Thanks for the update. I built a clean Fedora 34 with the same CC65 version installed via yum. Also installed byacc via yum. Did a fresh git clone of Fuzix. Went thru the make and make diskimage. Had two errors... both related to Fortune:

ERROR fortune:21 file not found
ERROR fortune.dat:23 file not found

After poking around, I used the older Makefile.6502 and the errors went to one, not finding the fortune.dat file. After examining the directory structure more, I noticed the fuzix-games.pkg file. It has a listing of files, but appears too small and won't extract. I decided to move it to the trash, went back to the new Makefile.6502 and the make diskimage now completes. It creates 3 files in the Image folder:

disk.img - 40.0MB
emu-ide.img - 40.0MB
filesys.img - 33.MB

Which img file would I use to build the compact flash image?

 Thanks again for looking into this... hoping to get the RC2014 cards built up for testing soon.

EtchedPixels, 2022-02-04:
disk is a raw CF image
emu-ide.img emulator
filesys.img just the file system itself