EtchedPixels/FUZIX #899: Question: Set TTY speed in Fuzix.

Spoozilla, 2022-01-24:
I have an RC2014 with an SC102 CTC and an SC104 SIO/2. I have decoupled the second SIO port from the system clock and I'm using one of the CTC channels to set the speed. In CP/M I have a small program that allows me to see a clock divider by issuing a couple of OUT commands. This allows me to connect an old RS232 terminal at 4800-8n1 to the machine. I'm trying to work out how achieve this from FUZIX. I've tried setting the channel in the monitor before booting but the SIO setup seems to reset the settings. I assume I can adjust the SIO initialisation code to issue these commands for the 2nd port but I'd rather find a way of doing it from the shell.

Is there a way of doing OUT commands from FUZIX?

EtchedPixels, 2022-01-24:
Nothing stops you writing an app that sends out instructions but the ideal way is to teach the serial code an option for your setup. How is the CTC/serial set up configured in your CP/M case ?

Spoozilla, 2022-01-24:
I have channel 1 of the CTC card input set to the system clock with the output set to CLK2. The second port of the SIO/2 has the clock input set to CLK2. I have a small compiled basic app that runs the following commands.

OUT &H89, &H55
OUT &H89, 12

This selects the selects the control register on channel 1 of the CTC and sets up counter mode, no interrupt, count on rising edge. It then selects the timer constant to produce a 3.072 Mhz signal which gets divided by 64 by the SIO/2 to produce 4800 baud.

EtchedPixels, 2022-01-25:
That configuration ought to work. The rc2014 code assumes that if a CTC is present and an SIO is detected that CLK2 is CTC output 1 and the SIO uses it. So

stty 4800 </dev/ttywhatever

or setting the baud rate in inittab for that line should work properly. I don't have any SIO in my systems any more to test however.

Spoozilla, 2022-01-25:
Unfortunately this doesn't work. I tried using stty previously, but I hadn't considered trying adding it to the inittab. However with either setting I get nothing. Even just dd'ing a text file to /dev/tty2 (without the 2nd terminal in inittab enabled) to the port does nothing and hangs before returning to the shell prompt. Ctrl-C does interrupt fine however.

Interestingly with CTC and SIO configured this way I can't connect at any speed to tty2, but if I return the sio2 2nd channel clock source to CLK I can connect at 115200 as usual.

EtchedPixels, 2022-01-25:
The relevant code is Kernel/platform-rc2014/devtty.c 

````
static void sio_do_setup(uint_fast8_t minor)
{
        struct termios *t = &ttydata[minor].termios;
        uint8_t r;
        uint8_t baud;
        uint8_t p = ttyport[minor];

        baud = t->c_cflag & CBAUD;
        if (baud < B300)
                baud = B300;

        /* Set bits per character */
        sio_r[1] = 0x01 | ((t->c_cflag & CSIZE) << 2);

        r = 0xC4;
        if (ctc_port && p == SIOB_C) {
                out(ctc_port + 1, 0x55);
                out(ctc_port + 1, siobaud[baud]);
                if (baud > B600)        /* Use x16 clock and CTC divider */
                        r = 0x44;
        } else if (z512_present && p == SIOB_C)
                z512_ctl = siobaud2[baud];
        else
                baud = B115200;

        t->c_cflag &= ~CBAUD;
        t->c_cflag |= baud;

        if (t->c_cflag & CSTOPB)
                r |= 0x08;
        if (t->c_cflag & PARENB)
                r |= 0x01;
        if (t->c_cflag & PARODD)
                r |= 0x02;
        sio_r[3] = r;
        sio_r[5] = 0x8A | ((t->c_cflag & CSIZE) << 1);
}
````

Which is meant to set up the CTC and the three relevant sio registers which the caller then writes to the chip.

Some kprintf() calls in there would show what paths and values were being used on your system

Spoozilla, 2022-01-27:
I put some debugging output in the code and everything looks to be in order. It selects the correct write register, it uses the right clock divider and the rest of the connection setup looks as expected when comparing the flags to the z80 users manual. Still nothing. So I thought I'd remove fuzix from the equation and use my known good CP/M configuration but with the values from devtty.c  and it doesn't work either. It seems the problem lies in the x16 clock mode. If I use the x64 clock mode and the correct CTC divider it works as expected. I should be able to test this with fuzix this afternoon.

My gut feeling is that maybe I have a faulty SIO chip here.

EtchedPixels, 2022-01-27:
More likely my setup values are wrong for those cases. I will take a look now you've confirmed the values are the problem.

Spoozilla, 2022-01-30:
This is turning into quite the journey. Who knew serial communication could get so complex?

CPM:
So, using the CTC and SIO in CP/M (both 2.2 and 3.0, with romwbw 3.0.1 and 3.1.1 1-pre.146) it turns out that any baud rate on the 2nd channel above 9600 results in dropped characters and gibberish. The primary channel for the main console works perfectly regardless at the usual 115200. Using the system clock for the 2nd SIO channel (resulting in 115200 there too) results in the same dropping and nonsense with the main console being unaffected. 

Fuzix:
Using the CTC and the SIO 2nd channel results in nothing, regardless of baud setting. The primary channel is perfect. With the SIO using the system clock for both channels (resulting in 115200) then both work flawlessly!?!

I've swapped between various FTDIs, rs232 and voltage converters, usb cables and terminals/servers. The only consistency is the inconsistency above....

I've been through the [z80 user guide](http://www.z80.info/zip/um0081.pdf) and confirmed a number settings. Which I can attach if you'd like to double check my workings. My next port of call is the romwbw source. I need to understand the differences between the two system's SIO setup routines. Why does Fuzix work at full system clock rates on both channels when CP/M does not? Why does the 2nd channel in CP/M fail at what seems like arbitrary clock rates? Why does x16 work in neither?

I can keep this thread updated but given the quite frankly odd nature of what's occurring here I completely understand if you'd rather close this issue.

EtchedPixels, 2022-01-30:
No .. it's most likely going to be something really silly that is worth getting to understand. Wondering if the CTC mode setup is wrong.

Spoozilla, 2022-10-31:
This is not a bug. It was 100% human error.

The CTC+SIO baud rate selection works perfectly. However I wasn't properly configuring the other end of the connection. I should have been ensuring that handshaking was enabled and flow control was off.

For Linux I ended up using -

`stty 115200 crtscts -ixon </dev/ttyUSB0`

This also allowed me to resolve the CP/M issue. 

```
OUT &H89, &H55 // control register on CTC channel 1. counter mode, no interrupt, count on rising edge.
OUT &H89, 2 // timer constant 2
OUT &H82, &H4 // select write register 4 on the SIO
OUT &H82, &H44 // x16 clock mode, 1 stop bit, no parity.
```

This provides a working x16 multiplied 2 clock divided signal for the 2nd SIO channel providing 115200 baud. When used in combination with the correct linux stty configuration above.

Sorry.

closed 2022-10-31 by Spoozilla