EtchedPixels/FUZIX #900: 6502 port using SDCC?

spth, 2022-02-08:
SDCC recently added support for the 6502 (it is currently still considered "in development", but already passes nearly all regression tests).

It might make sense to check if the SDCC mos6502 port is useable for FUZIX.

EtchedPixels, 2022-02-08:
Possibly - cc65 lacks float and the code generation could be better. It's also however extremely reliable.Something to look at in a couple of years maybe when/if SDCC on 6502 is stable.

NollKollTroll, 2022-02-20:
There is also a promising LLVM/Clang/6502 in the works.
https://github.com/mysterymath/clang6502
This is quite interesting reading: 
http://forum.6502.org/viewtopic.php?f=2&t=6450&sid=99a4fa3644c432d5a4c7e1778ddb9b1a

erkinalp, 2022-02-20:
>  https://github.com/mysterymath/clang6502

404 :(

NollKollTroll, 2022-02-20:
Sorry, this should be the correct link:
https://github.com/llvm-mos/llvm-mos

spth, 2022-02-21:
AFAIK, there is also a GCC port. And more C compilers targeting 6502. The author of the SDCC port has posted a benchmark comparison (according to which sdcc looks quite good performance wise):

https://sourceforge.net/p/sdcc/mailman/message/37614056/

However, I expect all of them to still have some stability issues at this point. Still, it will be interesting to see how this evolves (both performance- and correctness-wise) over the next few years.

EtchedPixels, 2022-02-21:
It's also not just about stability and speed. The SDCC assembler/linker setup is just horrible compared to cc65 so any move would involve a lot of hackery and writing tools to get usable binaries and toolchains. As it is if I had the time I'd extend the Fuzix Z80 assembler and linker to speak SDCC's weird Z80 dialect and just use the compiler phase.