EtchedPixels/FUZIX #907: Trouble building games for 6502

NollKollTroll, 2022-02-20:
I base my own platform PZ1 on rc2014-6502 and since getting the latest source, I have trouble building.

Steps to reproduce on Linux Mint 20.3 with cc65 from a 2 days ago:
- create a git clone. Also tested with the ZIP from github.
- 'make clean'
- changing the platform in FUZIX/makefile to rc2014-6502
- 'make'
- 'make diskimage' fails with:
Processing package flashrom
Processing package games
ERROR fortune:21 file not found
ERROR fortune.dat:23 file not found
Processing package ld09
Processing package ue
Processing package util-mini
Processing package util
ERROR Fix errors and re-run
make: *** [Makefile:115: diskimage] Error 2

Details of what I can find:
- 'make diskimage' does not create fortune, fortune.dat or fortune-gen-linux.
- 'make clean' fails to remove fortune, fortune.o, fortune.dat and fortune-gen-linux when they are there.
- fortune.c should not have to be rebuilt if fortune.dat is changed?
- size.report does not include fortune.dat.
- fortune-gen-linux tries to use cc65, so I used gcc specifically. I am probably missing some env-setting?
I've since gotten it to create something, but am unsure about what is correct and why things are done the way they are.
Looking at the other platforms and older revisions of the file I now have the following make.6502 that works for me. I have no rc2014-6502 to test it on.

`include $(FUZIX_ROOT)/Applications/rules.6502

SRCSNS = fortune.c qrun.c advint.c z1.c z2.c z3.c z4.c z5.c z8.c

SRCS  = adv01.c adv02.c adv03.c adv04.c adv05.c adv06.c adv07.c \
        adv08.c adv09.c adv10.c adv11.c adv12.c adv13.c adv14a.c adv14b.c \
        myst01.c myst02.c myst03.c myst04.c myst05.c myst06.c myst07.c \
        myst08.c myst09.c myst10.c myst11.c fortune-gen.c startrek.c \
	hamurabi.c cowsay.c taylormade.c

OBJS = $(SRCS:.c=.o)
OBJSNS = $(SRCSNS:.c=.o)

APPS = $(OBJS:.o=)
APPSNS = $(OBJSNS:.o=)

CFLAGS += -DCONFIG_IO_CUSS

all: $(APPS) $(APPSNS) fortune.dat size.report

$(APPS): %: %.o
	$(LINKER) -o $@ $(LINKER_OPT) $(CRT0) $^ $(LINKER_TAIL)

$(APPSNS): %: %.o
	$(LINKER) -o $@ $(LINKER_OPT) $(CRT0NS) $^ $(LINKER_TAIL)

size.report: $(APPS) $(APPSNS) fortune.dat
	ls -l $^ > $@

clean:
	rm -f $(OBJS) $(OBJSNS) $(APPS) $(APPSNS) $(APPTC) $(SRCS:.c=) \
        core *~ *.asm *.lst *.sym *.map *.noi *.lk *.ihx *.tmp *.bin \
        fortune.dat fortune-gen-linux size.report

rmbak:
	rm -f *~ core

#
#	Build the fortune data file
#
fortune-gen-linux: fortune-gen.c
	gcc -o fortune-gen-linux $<

fortune.dat: fortune-gen-linux
	./fortune-gen-linux < fort.db >fortune.dat

#fortune: fortune.dat fortune.o
#	$(LINKER) -o $@ $(LINKER_OPT) $(CRT0NS) fortune.o $(LINKER_TAIL)`

EtchedPixels, 2022-02-20:
I thought I'd fixed all of that but there were some bits missing. I'm trying to get it all to the point enough is common I can have a tool that turns a skeleton list into a set of Makefiles to end the whole makefile maintenance nightmare but not there yet.

Seems to work for me now, will push in a moment

NollKollTroll, 2022-02-20:
Just tested latest version. It builds OK here now.
The size.report is missing fortune and fortune.dat, only fortune-gen is reported.
'make clean' does not remove fortune.o
I'm happy to move along, trying to get a disk-image read from SD :D

EtchedPixels, 2022-02-27:
Fixed the other glitches I found

closed 2022-02-27 by EtchedPixels