EtchedPixels/FUZIX #908: Esp8266 diskimage error

kazueda, 2022-03-06:
Dear Sir,
I am trying to boot a recent fuzix filesystem on an esp8266 board.
I succeed in compling  two ".bin" file with the command "make TARGET=esp8266"
But I can not make diskimage with an error message as follows.
I use an xtensa-lx106-elf-gcc (9.2.1-29ubuntu1+6) 9.2.1 2020022020.04 on an ubuntu 20.04 machine. 

mkdir -p Images/esp8266
(cd Standalone/filesystem-src; ./build-filesystem  /root/FUZIX_esp8266_220306/Images/esp8266/filesys.img 256 65535)
Found package mini
Found package basefs
Found package rpilot
Found package V7-games
Found package V7-games-man
Found package V7-cmd-micro
Found package V7-cmd
Found package V7-cmd-man
Found package V7-sh
Found package V7-fsh
Found package V7-sh-man
Found package fforth
Found package util-mini
Found package util
Found package BCPL
Found package TCL
Found package ue
Found package cursesgames
Found package as09
Found package games
Found package ld09
Found package flashrom
Found package drivewire
Found package ccp
Found package SmallC
Found package levee-vt52
Found package levee-ansi
Found package levee-termcap
Found package dasm09
Found package adventure
Found package netd-base
Found package netd-lwwire
Found package netd-coconic
Found package netd-slip
Found package netd-eth
Found package MWC-cmd
Found package MWC-float
Found package fview
Found package platform-dk-tm4c129x
Found package platform-dragon-nx32
Resolving package mini: enabled=0
Resolving package basefs: enabled=1
Resolving package rpilot: enabled=0
Resolving package V7-games: enabled=1
Resolving package V7-games-man: enabled=1
Resolving package V7-cmd-micro: enabled=1
Resolving package V7-cmd: enabled=1
Resolving package V7-cmd-man: enabled=1
Resolving package V7-sh: enabled=1
Resolving package V7-fsh: enabled=1
Resolving package V7-sh-man: enabled=1
Resolving package fforth: enabled=1
Resolving package util-mini: enabled=1
Resolving package util: enabled=1
Resolving package BCPL: enabled=0
Resolving package TCL: enabled=0
Resolving package ue: enabled=1
Resolving package cursesgames: enabled=1
Resolving package as09: enabled=1
Resolving package games: enabled=1
Resolving package ld09: enabled=1
Resolving package flashrom: enabled=1
Resolving package drivewire: enabled=0
Resolving package ccp: enabled=0
Resolving package SmallC: enabled=1
Resolving package levee-vt52: enabled=0
Resolving package levee-ansi: enabled=0
Resolving package levee-termcap: enabled=0
Resolving package dasm09: enabled=0
Resolving package adventure: enabled=1
Resolving package netd-base: enabled=1
Resolving package netd-lwwire: enabled=0
Resolving package netd-coconic: enabled=0
Resolving package netd-slip: enabled=0
Resolving package netd-eth: enabled=0
Resolving package MWC-cmd: enabled=1
Resolving package MWC-float: enabled=1
Resolving package fview: enabled=0
Resolving package platform-dk-tm4c129x: enabled=0
Resolving package platform-dragon-nx32: enabled=0
Processing package basefs
Processing package V7-games
Processing package V7-games-man
Processing package V7-cmd-micro
Processing package V7-cmd
ERROR clear:12 file not found
Processing package V7-cmd-man
Processing package V7-sh
Processing package V7-fsh
Processing package V7-sh-man
Processing package fforth
Processing package util-mini
Processing package util
Processing package ue
Processing package cursesgames
Processing package as09
Processing package games
ERROR taylormade:43 file not found
Processing package ld09
Processing package flashrom
Processing package SmallC
Processing package adventure
Processing package netd-base
Processing package MWC-cmd
Processing package MWC-float
ERROR Fix errors and re-run
make: *** [Makefile:115: diskimage] Error 2

Sincerely yours,
kazueda

EtchedPixels, 2022-03-07:
Thanks. Will take a look. Probably a couple of makefiles have ended up out of date.

EtchedPixels, 2022-03-07:
Should be fixed

closed 2022-03-07 by EtchedPixels