EtchedPixels/FUZIX #915: A possible 65816 Compiler called "Calypsi C"

ProxyPlayerHD, 2022-05-28:
it seems that currently this is just using cc65 with some hacks.
and i just wanted to ask if anyone on here has taken a look at [Calypsi C](https://github.com/hth313/Calypsi-tool-chains)?
it's a C99 Compiler for the 65816 (and other CPUs) with both `float` and 64-bit data type support like `long long` and `double`.

i did kind of try to compile the whole thing using Calypsi C because i recently got it set up and running, so was still riding that high and felt confident.
but i'm a Windows pleb, so i don't even know where to begin converting all the makefile stuff to batch or just trying to find all mentions of `cc65` and replace them with `cc65816`, and rewrite all the assembly to ca65816's syntax.

so i think i'll just leave that to the smart people on here and just point out: "hey, there is a relatively new 65816 compiler, if you haven't already check it out and see if it would work for this project"
honestly i wish i could help more than that.

EtchedPixels, 2022-05-28:
From the website

"The tools are closed source."

so not really an option.

closed 2022-05-28 by EtchedPixels

ProxyPlayerHD, 2022-05-28:
aww that's a shame.
i assume the dislike of closed source is because when the devs stop working on it you're stuck with a black box that can never be updated again or even have it's internals reused?
i guess that's also the reason why WDC's own Compiler isn't an option either, i mean it's basically just abandonware at this point.
despite that i doubt they would release the source code when asked, though i'm not sure if anyone did ever ask.

oh well, maybe after cc65 is more matured and got all the features the devs wanted to add, some of them will start on a version for the 65816. i don't know if that's on their roadmap or way too far into the future to plan for.

EtchedPixels, 2022-05-29:
Not really - the 65C816 in 16bit mode is a very different processor.

ProxyPlayerHD, 2022-05-29:
that's not what i meant, "port" was a poor choice of words on my part.
i meant that the cc65 team could make a new compiler from scratch specifically for the 65816.
and it wouldn't need to be a seperate repo or anything, seeing as the linker, assembler, and librarian can already handle the 65816, so it could just be added to the existing cc65 utility.
but again i don't know if that is on their roadmap or just an idea for `Later™`

EtchedPixels, 2022-05-30:
What I meant by the 16bit mode being very different affects OS, applications and compilers. From a programming perspective the 65C816 in native mode is totally different because you've got 16bit maths, a proper (although somewhat messed up) stack, movable zero page and so on.

I'd also be surprised if cc65 ever did as 65C816 port because it would be a massive piece of work. There is a tcc port for 65C816. It's used by the SNES SDK, not sure what state it is in.

Even running mostly 8bit mode my 65C816 board is plenty fast enough so its not high on my list at least.