EtchedPixels/FUZIX #916: fsck corrupting floppy disk filesystem

xmotif, 2022-05-31:
I have been using the EtchedPixels Floppy Controller with my Z80 RC2014. In FUZIX (latest git), I can run mkfs on a floppy successfully, but when I run fsck it destroys the first block after the inodes.

For debugging, I added two lines to fd_transfer in Kernel/dev/devfd.c:
````
       	for (;;)
       	{
                irq = di();
-               if (rwflag)
+               if (rwflag) {
+                       kprintf("fd: trk%u sec%u rd\n", (unsigned int)devfd_track, (unsigned int)devfd_sector);
                       	retc = devfd_read(minor);
-               else
+               } else {
+                       kprintf("fd: trk%u sec%u wr\n", (unsigned int)devfd_track, (unsigned int)devfd_sector);
                       	retc = devfd_write(minor);
+               }
               	irqrestore(irq);
````

<details><summary>Creating filesystem with mkfs, success (click to open):</summary>

````
# mkfs /dev/fd0 32 180
Making filesystem on device /dev/fd0 with isize 32 fsize 180. Confirm? y
fd: trk0 sec0 wr
fd: trk0 sec1 wr
fd: trk0 sec2 wr
fd: trk0 sec3 wr
fd: trk0 sec4 wr
fd: trk0 sec5 wr
fd: trk0 sec6 wr
fd: trk0 sec7 wr
fd: trk0 sec8 wr
fd: trk0 sec9 wr
fd: trk0 sec10 wr
fd: trk0 sec11 wr
fd: trk0 sec12 wr
fd: trk0 sec13 wr
fd: trk0 sec14 wr
fd: trk0 sec15 wr
fd: trk0 sec16 wr
fd: trk0 sec17 wr
fd: trk1 sec0 wr
fd: trk1 sec1 wr
fd: trk1 sec2 wr
fd: trk1 sec3 wr
fd: trk1 sec4 wr
fd: trk1 sec5 wr
fd: trk1 sec6 wr
fd: trk1 sec7 wr
fd: trk1 sec8 wr
fd: trk1 sec9 wr
fd: trk1 sec10 wr
fd: trk1 sec11 wr
fd: trk1 sec12 wr
fd: trk1 sec13 wr
fd: trk1 sec14 wr
fd: trk1 sec15 wr
fd: trk1 sec16 wr
fd: trk1 sec17 wr
fd: trk2 sec0 wr
fd: trk2 sec1 wr
fd: trk2 sec2 wr
fd: trk2 sec3 wr
fd: trk2 sec4 wr
fd: trk2 sec5 wr
fd: trk2 sec6 wr
fd: trk2 sec7 wr
fd: trk2 sec8 wr
fd: trk2 sec9 wr
[... 354 lines cut ...]
fd: trk1 sec4 rd
fd: trk1 sec5 rd
fd: trk1 sec6 rd
fd: trk1 sec7 rd
fd: trk1 sec8 rd
fd: trk1 sec9 rd
fd: trk1 sec10 rd
fd: trk1 sec11 rd
fd: trk1 sec12 rd
fd: trk1 sec13 rd
fd: trk0 sec2 wr
fd: trk1 sec14 wr
fd: trk0 sec1 rd
Device 0 has fsize = 180 and isize = 32. Continue? y
Pass 1: Checking inodes...
fd: trk0 sec2 rd
Inode 1 with mode 010033 is not of correct type. Zap? y
Inode 2 with mode 0146705 is not of correct type. Zap? y
Inode 3 with mode 0156614 is not of correct type. Zap? y
Inode 4 with mode 074526 is not of correct type. Zap? fd: trk0 sec2 wr
Interrupt
# 
````

</details

EtchedPixels, 2022-05-31:
I will try and take a look when I get some time.

xmotif, 2022-06-01:
I narrowed down the source of the track1,sector14 write during the first fsck after filesystem creation. It is during pass 4 of fsck, ckdir function, in line 594 of fsck-fuzix.c: `dirwrite(&ino, j, &dentry);`
I haven't yet learned enough about the filesystem to determine what's going on here.

EtchedPixels, 2022-06-02:
ckdir walks all of the directories making sure that they make sense. It scrubs any bytes after the \0 end marker to zero and it's not smart enough to not write it back if this did nothing. That said it should still be writing back valid data and that doesn't explain why only the FDC breaks.

xmotif, 2022-06-02:
I have a Greaseweazle on the way so I can rule out any hardware issues and verify what's really on the disk

xmotif, 2022-06-03:
I did a thorough test with commit 7ad2cdc181f8706dfee5105fd1d3a017db07d290 and it fixed all of the problems. Thank you!

commit 7ad2cdc1 (2022-06-02): devfd: set the user/kernel flag back for a kernel copy

Hopefully fixes #916 but not verfified
closed 2022-06-03 by xmotif