EtchedPixels/FUZIX #922: Raspberry Pi Pico System Error

andrei-edward-popa, 2022-06-24:
Hi! I want to load Fuzix OS on as Raspberry Pi Pico. I build Fuzix for Pico and fuzix.uf2 file was generated successfully. After that, I copid the file to the Pico partition and I tried to run ls command, but I got the following output:

FUZIX version 0.4pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2022 Alan Cox <alan@etchedpixels.co.uk>
Devboot
264kB total RAM, 160kB available to processes (15 processes max)
Enabling interrupts ... ok.
NAND flash, 1952kB physical 1296kB logical at 0x13018000: hda: 
SD drive 0: no card found
Mounting root fs (root_dev=0, ro): warning: mounting dirty file system, forcing r/o.
OK
Starting /init
init version 0.9.0ac#1
Cannot open file
Current date is Thu 2022-06-23
Enter new date: 
Current time is 19:45:32
Enter new time: 

 ^ ^
 n n   Fuzix 0.3.1
 >@<
       Welcome to Fuzix
 m m

login: root

Welcome to FUZIX.
# ls
FLAGRANT SYSTEM ERROR! EXCEPTION 3
 r0=2001BEE6 r1=2001BFDA  r2=00000001  r3=2001C118
 r4=2001BF44 r5=2001C128  r6=2001AB10  r7=2001C144
 r8=00000001 r9=2001628C r10=2001BF08 r11=2001689D
r12=2001A139 sp=2001BD78  lr=200188A3  pc=20016E22
PROGBASE=2001628C PROGLOAD=2001688C PROGTOP=2002628C
UDATA=2001628C KSTACK=2001639C-2001688C
user mode relative: lr=00002017 pc=00000596 isp=2001BEA0 brk=2001C61C

panic: fatal exception

I thought that my Pico had some sort of problem and I tried with another, but I got more confusion because with the new board I got the following:

FUZIX version 0.4pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2022 Alan Cox <alan@etchedpixels.co.uk>
Devboot
264kB total RAM, 160kB available to processes (15 processes max)
Enabling interrupts ... ok.
NAND flash, 1952kB physical 1277kB logical at 0x13018000: hda: 
SD drive 0: no card found
Mounting root fs (root_dev=0, ro): failed

panic: no root

Is this a known issue or it is something that I miss?

andrei-edward-popa, 2022-06-24:
@EtchedPixels Ok, I understand, do you know where I can find this patch? Also, does it work to run commands without a filesystem on a SD Card? Because I don't have an SD card reader right now.

EtchedPixels, 2022-06-24:
You can run it without using the internal flash fs. I don't know why your second one isn't detected the internal fs if you did set it up that way though.

davidgiven, 2022-07-04:
Chances are that the crash is due to a required sh fix, which has now been merged. Try updating and rebuilding everything.

TurBoss, 2022-07-08:
up and running on nand flash

```
FUZIX version 0.4pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2022 Alan Cox <alan@etchedpixels.co.uk>
Devboot
264kB total RAM, 160kB available to processes (15 processes max)
Enabling interrupts ... ok.
NAND flash, 1952kB physical 1296kB logical at 0x13018000: hda: 
SD drive 0: no card found
Mounting root fs (root_dev=0, ro): warning: mounting dirty file system, forcing r/o.
OK
Starting /init
init version 0.9.0ac#1
Cannot open file
Current date is Fri 2022-07-08
Enter new date: 
Current time is 14:26:17
Enter new time: 

 ^ ^
 n n   Fuzix 0.3.1
 >@<
       Welcome to Fuzix
 m m

login: root

Welcome to FUZIX.
# pwd
/root
# whoami
root
# 
```

now I will try SD

this is on dtrg-esp8266 e85ac31c9427e8efa048f113535009200f71df51

thanks!

closed 2022-07-17 by EtchedPixels