EtchedPixels/FUZIX #931: Creating CF disk image for custom computer

ddraig68k, 2022-07-14:
I'm currently trying to port FUZIX to my 68000 based computer. I have the kernel built and it seems to be running but I seem to get getting problems when booting from the disk.

I'm currently loading the kernel into RAM and executing it. When running i'm getting up to the point where it's trying to access the disk:
```
A:/> loadmem fuzix.bin 100000
Loaded 42724 bytes from file fuzix.bin to location 100000
A:/> run 100000
FUZIX version 0.4pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2022 Alan Cox <alan@etchedpixels.co.uk>
Devboot
Motorola 68000 processor detected.
8196kB total RAM, 7094kB available to processes (64 processes max)
Enabling interrupts ... ok.
IDE drive 0: hda: hda1 hda2 hda3
IDE drive 1: ide error, status=0000
bootdev:
```
No matter which partition I select I get the same error message.
```
bootdev: hda
Mounting root fs (root_dev=0, ro): failed
bootdev: hda1
Mounting root fs (root_dev=1, ro): failed
bootdev: hda2
Mounting root fs (root_dev=2, ro): failed
bootdev: hda3
Mounting root fs (root_dev=3, ro): failed
```

The hda3 partition is a FAT32 partition which is required by my current DOS system so I can load the kernel into RAM.

I'm using the following to make the disk image file:
```
diskimage:
	# Make a blank disk image with partition
	dd if=$(FUZIX_ROOT)/Standalone/filesystem-src/parttab.40M of=$(IMAGES)/disk.img bs=40017920 conv=sync
	# Add the file system
	dd if=$(IMAGES)/filesys.img of=$(IMAGES)/disk.img bs=512 seek=2048 conv=notrunc
	cp ../fuzix.bin $(IMAGES)
```
I'm building FUZIX using WSL on Windows then using [Rufus](https://rufus.ie/en/) to write the disk.img to the CF card first, then creating the FAT32 partition after that.

Is there anything obvious that i'm doing wrong here? Is this the correct way to create a disk image? Failing that, is there any sort of debug mode that I can enable to check out what's happening when it's trying to read the disk?

ddraig68k, 2022-07-15:
Not sure if this helps. after defining DEBUG I get the following when trying to boot from disk:
```
bootdev: hda1
Mounting root fs (root_dev=1, ro): fp->s_mounted=0x0000, fp->s_isize=0x00FF, fp->s_fsize=0x1033 failed
```
I suspsect that it could be down to how the disk image is created possibly how it's being written to the disk.

EtchedPixels, 2022-07-16:
It's finding and doing the basic disk I/O ok as it is reading the partitions. However from the dump it looks like the partitions don't actually contain the expected file system data. That would suggest something went wrong setting the disk image up. You added a partition and stuff - does hda1 still start at block 2048 ?

ddraig68k, 2022-07-16:
The script has seek=2048 so it should be starting at that block. Does it require the partition to start at that location? I have tried changing it so the parition starts at the beginning of the CF card as well but i get the exact same issue happen. 

Will try creating the disk under Linux rather than windows and see if I get the same issue.

EtchedPixels, 2022-07-16:
The dd is putting the file system 2048 blocks in and my original partition table has the first partition there. I've no idea how this all works out with Windows I'm afraid or whether rufus went and created a new set of partitions and the like.

ddraig68k, 2022-07-17:
I seem to be making some progress here. Writing the image file to the card using Linux seems to work better. The fp->s_mounted value was byte swapped so I added the *swab* to the dd command line the same as on the tiny68k target. This time the disk seemed to mount at least but it's failing to find /init.
```
Motorola 68000 processor detected.
8196kB total RAM, 7094kB available to processes (64 processes max)
Enabling interrupts ... ok.
IDE drive 0: hda: hda1 hda2
IDE drive 1: ide error, status=0000
bootdev: hda1
Mounting root fs (root_dev=1, ro): fp->s_mounted=0x31C6, fp->s_isize=0x0100, fp->s_fsize=0xFFFF
OK
Starting /init

panic: no /init
```

I tried the 68000 image file from the fuzix website and that seems to get a little further before the computer halts. I'm assumming that's because it's for a different target machine than mine.
```
Devboot
Motorola 68000 processor detected.
8196kB total RAM, 7094kB available to processes (64 processes max)
Enabling interrupts ... ok.
IDE drive 0: hda: hda1 hda2
IDE drive 1: ide error, status=0000
bootdev: hda1
Mounting root fs (root_dev=1, ro): fp->s_mounted=0x31C6, fp->s_isize=0x0100, fp->s_fsize=0xFFFF
OK
Starting /init
init version 0.9.0ac#1
```

Now I need to work out what's happening with the file system.

ddraig68k, 2022-07-17:
It's working now building and from the disk. I hadn't set the ENDIANFLAG value. Still crashing on init but that's a separate issue.

closed 2022-07-17 by ddraig68k

EtchedPixels, 2022-07-17:
Good to know. The swab on the Tiny68K is because the interface is hardware byteswapped by mistake.