EtchedPixels/FUZIX #935: DS12885

xmotif, 2022-08-06:
Hi, I built a DS12885 board because I agree with the advantages stated on the [Hackaday page](https://hackaday.io/project/184734-rc2014-pc-style-rtc-ds12885)

Right now it looks like there is just ds12885.c/.h in the Kernel/dev directory. Is there perhaps some other unpublished prototype code? I would like to implement the ds12885 for RC2014, but first want to see if there is already some work-in-progress that I could build on.

EtchedPixels, 2022-08-07:
That's the prototype code. I've not had time to then wire it up to stuff except test code.

xmotif, 2022-08-08:
I've been working on it and may submit a PR this week, at the minimum for RTC function. But I would like to eventually get the interrupt timer working as well, to remove the CTC.

xmotif, 2022-09-21:
I have some basic code to hook it up, but the DS12885 is a less than ideal timer interrupt. 64 Hz ticks are missed during disk access, causing the system time to drift. The only solution is to read the RTC during every time() system call. I still have bugs to work out.

xmotif, 2022-09-21:
Problem with time drift has been solved by correcting logic error in plt_rtc_secs. Solving a regression with floppy driver then will submit PR.

xmotif, 2022-10-11:
#941 fixes an issue where timer_source was being reset back to CTC every time the floppy was used.

xmotif, 2022-11-17:
PR #942 opened for DS12885 support

EtchedPixels, 2022-11-18:
Thanks - looks good

closed 2022-11-18 by xmotif