EtchedPixels/FUZIX #938: rc2014-6502 emulator: missing 6502.rom file

piso77, 2022-09-26:
[rc2014-6502](https://github.com/EtchedPixels/FUZIX/tree/master/Kernel/platform-rc2014-6502/README) refers to a '6502.rom' file to be downloaded from the [RC2014 ROM archive](https://github.com/EtchedPixels/RC2014-ROM.git) but none is found there (and building the 6502 subdit won't produce one).

NollKollTroll, 2022-09-27:
I've successfully done what you try to do a week ago, so to make sure it still works I now did a clean make on Fuzix, RC2014-ROM and RC2014. I got the emu working with the following command while being in the RC2014 folder:
./rc2014-6502 -i ../FUZIX/Images/rc2014-6502/emu-ide.img -r ../RC2014-ROM/6502/rom
So at least for me it really does produce the required rom-file, and it does work in the emu. The rom-file is 512KiB in size.

piso77, 2022-09-30:
It worked, README fix submitted:

https://github.com/EtchedPixels/FUZIX/commit/e36af07eea6f5df1df41c86a07fd0753fccb99d7

closed 2022-09-30 by piso77