EtchedPixels/FUZIX #948: cat allows reading of directories

Darin755, 2022-12-11:
It seems like cat doesn't check to see if the argument provided is a file and not a directory.

When I run ```cat /``` on my RPI Pico I get the following:
```...3bin2dev1etc0mnt/root.tmp-usr,varinit```

Ideally cat should print a error instead of trying to read it like a file. I would be ideal if cat gave a message like:
```cat: /: Is a directory``` (Error from GNU core utils on linux)

erkinalp, 2022-12-11:
As fuzix only intends to support a single file system format (per closed #932), this should be acceptable. GNU does not have that luxury.

EtchedPixels, 2022-12-11:
cat never has checked. On traditional Unix like systems a directory is a readable file although the contents are file system dependent. The fact it gives you an error on the GNU utils in Linux is because Linux doesn't usually allow the reading of a directory except via special calls.

If you ran the same GNU tools on says a System V.2 Unix box you'd get the Fuzix behaviour.

As such I don't think this is a bug.

closed 2022-12-11 by EtchedPixels