EtchedPixels/FUZIX #954: Trouble compiling sc126 with Wiznet support

mennozon, 2023-01-03:
The README mentions Wiznet 5500 on SPI can be used as an option but I'm unable to compile the kernel if I enable this in config.h

I'm using Debian (tested on both Bullseye and Bookworm) with [sdcc280](https://github.com/EtchedPixels/sdcc280), I assume this is still needed because of the mentioned bugs in #901 and wiki.

Is sc126 with Wiznet w5500 'supported' or possible at this time?

I assume the kernel gets too big when enabling Wiznet but I'm not sure where to go from there.

> head -5 hogs.txt
> 1500: _i_tab
> 1405: _netproto_sockpending
> 1291: __execve
> 1095: _ptab
> 1048: _syscall_dispatch
> tools/visualize < fuzix.map
> #000000000000000 0000000000000222 2222222222222222 2CIDDDDBBiXXSSS# 
> 0000000000000000 0000000000000222 2222222222222222 2CDDDDDBBiXXSSS# 
> 0000000000000000 0000000000000222 2222222222222222 2CDDDDDBBXXXSS## 
> 0000000000000000 0000000000000222 2222222222222222 CIDDDDDBBXX#SS## 
> 
> 
> tools/bihx fuzix.ihx
> tools/binprep
> Code1: 29116 bytes
> Code2: 20465 bytes
> Code:  49581 bytes
> Data:  5242 bytes
> BSS:   0 bytes
> Initialized: 247 bytes
> Free memory begins at: e47f
> Common is at: f000
> Space: 2945 bytes
> Work room: 2698 bytes
> Initializer will have overwritten DISCARD
> make[1]: *** [cpu-z180/image.mk:33: fuzix.bin] Error 1

EtchedPixels, 2023-01-03:
At the moment you need to drop something else to make it all fit. It's on the long todo list to finish supporting banking networking on Z180

mennozon, 2023-01-04:
I was able to make it fit by disabling i2c and sdcard support, FUZIX now detects the Wiznet device and it is working, thanks!

closed 2023-01-04 by mennozon