EtchedPixels/FUZIX #960: fuzixlod for platform-v8080

kazueda, 2023-01-27:
Dear Sir,
I am trying to make a fuzix.bin for the v8080 platform.
In the "fuzix" file for the ACK compiler, there is a notation of  "program {EM}/bin/fuzixlod."
When I execute "make TARGET=v8080", my computer said as follows.

ack -mfuzix cpuinfo-8080.s
sh: 1: /usr/local/bin/fuzixlod: not found
make[2]: *** [Makefile.8080:145: cpuinfo-8080.o] Error 1
make[2]: Leaving directory '/root/FUZIX_v8080_221218/Applications/util'
make[1]: *** [Makefile:10: util] Error 2
make[1]: Leaving directory '/root/FUZIX_v8080_221218/Applications'
make: *** [Makefile:106: apps] Error 2

How can I obtain the "fuzixlod" file ?
Sincerely yours,

EtchedPixels, 2023-01-28:
I think it's also in the Fuzix tree as Library/tools/ack2fuzix.c

EtchedPixels, 2023-06-29:
Now switched to the new compiler so no longer relevant or needed so closing

closed 2023-06-29 by EtchedPixels