EtchedPixels/FUZIX #966: Test current sdcc (before 4.3.0 release)

spth, 2023-02-04:
SDCC developers intend to do a 4.3.0 release soon. While the official release process with code freezes, etc hasn't been started yet, this seems like a good time to test if the FUZIX fork of SDCC works well with SDCC from current svn, and report any found SDCC issues upstream, while there is still a chance to fix them before 4.3.0.

SDCC progress since 4.2.0 was mostly in improved standard compliance; a large number of bugs have been fixed, too.

EtchedPixels, 2023-02-04:
I'm still on the old calling API and have not ported the banking patches to a modern SDCC. I don't currently plan to move versions because of the huge amount of work involved.

spth, 2023-02-04:
Could switching to current sdcc, but using the old calling convention (via --sdcccall 0) be a way forward for FUZIX?

EtchedPixels, 2023-02-22:
For banked it would have to be I think because you need some free registers to bank switch nicely.

Closing this though as there's simply no way it will happen before any 4.3 release

closed 2023-02-22 by EtchedPixels