EtchedPixels/FUZIX #976: Missing some functions for pdp11 build

merclangrat, 2023-03-23:
I am trying to build FUZIX pdp11 kernel and get the following while `make kernel`:
`pdp11-aout-ld -L../../Library -M -o fuzix.aout -T fuzix.ld \
crt0.o \
../start.o ../version.o ../lowlevel-pdp11.o \
main.o ../swap.o ../timer.o ../simple.o ../kdata.o devices.o \
../tty.o ../devio.o ../filesys.o ../process.o ../inode.o ../syscall_fs.o \
pdp11.o ../syscall_proc.o ../syscall_other.o ../mm.o \
../devsys.o ../usermem.o ../syscall_exec16.o ../syscall_fs2.o \
tricks.o ../syscall_fs3.o \
../usermem_std-pdp11.o devtty.o libc.o > ../fuzix.map
pdp11-aout-ld: ../filesys.o:../filesys.o:(.text+0x782): undefined reference to `breadi'
pdp11-aout-ld: ../filesys.o:../filesys.o:(.text+0x814): undefined reference to `inode_blocks'
pdp11-aout-ld: ../filesys.o:../filesys.o:(.text+0x832): undefined reference to `bmap'
pdp11-aout-ld: ../filesys.o:../filesys.o:(.text+0xd6a): undefined reference to `bwritei'
pdp11-aout-ld: ../inode.o:../inode.o:(.text+0x3e): undefined reference to `bmap'
pdp11-aout-ld: ../syscall_exec16.o:../syscall_exec16.o:(.text+0x192): undefined reference to `rargs'
pdp11-aout-ld: ../syscall_exec16.o:../syscall_exec16.o:(.text+0x306): undefined reference to `wargs'
make[2]: *** [Makefile:32: image] Error 1`

I ran `make ltools` and `make libs` before. I can't find those functions in the repository, may I need an external library to be built?

merclangrat, 2023-03-23:
When I added to this like `blk400.o`, `blk512.o` and `syscall_exec.o`, I get other error:
`tricks.o:tricks.o:(.text+0x98): relocation truncated to fit: DISP16 against symbol `outr0hex' defined in .text section in ../lowlevel-pdp11.o`
`../syscall_fs3.o:../syscall_fs3.o:(.text+0x502): relocation truncated to fit: DISP16 against symbol `pagemap_mem_used' defined in .text section in ../simple.o`

I guess I can try to find a solution (and create a PR/diff) but would be nice if someone already knows smth about that

EtchedPixels, 2023-03-23:
That's where the port got to - compiler bugs and binutils bugs so I gave up. The compiler bugs are now fixed it seems the binutils not.

merclangrat, 2023-03-23:
@EtchedPixels Ah, I see, thank you for the explanation. if I am able to do something, I create a PR

https://github.com/sergev/bkunix - this project uses another compiler (BK is a Soviet computer with LSI-11 compatible CPU, my idea is to try FUZIX on BK), they say too that it didn't go so well with gcc: `An initial attempt to use GCC for cross-development failed: it produced a buggy and size-inefficient code`
then I totally see your point, thanks again!

EtchedPixels, 2023-03-25:
It actually looks like the binutils problems left are wrong warnings. It's confused and doesn't seem to understand that a 16bit PC relative relocation on a PDP11 wraps so is never out of range.

EtchedPixels, 2023-03-26:
Had a poke at it - with gcc 12.2.0 and binutils 2.40 it seems to be generating believable output except that the linker wrongly assumes that wrapping on 16bit offsets is bad.

If you set the bfd/pdp11.c entries in binutils to complain_overflow_dont then it seems to be a possibly working toolchain. Ragge's ANSI pcc 1.1 would be another possibility - but would need a suitable assembler/linker and I don't know what it assumes/requires.

merclangrat, 2023-03-28:
> Had a poke at it - with gcc 12.2.0 and binutils 2.40 it seems to be generating believable output except that the linker wrongly assumes that wrapping on 16bit offsets is bad.
>
> If you set the [...]

Thank you very much! Yes, it worked, it's enough for me to proceed

closed 2023-03-28 by merclangrat