EtchedPixels/FUZIX #982: zxdiv with DivMMC Future and +3: "init not found"

ninakali, 2023-03-30:
Hello! I've been trying to make fuzix work on my +3 for a while. I have DivMMC Future running EsxDos 0.87.

I've tried kernels built from master (today) and [600f2cbc0ecbc01fbb5ca6c825fbabfa85f652f1](https://github.com/EtchedPixels/FUZIX/commit/600f2cbc0ecbc01fbb5ca6c825fbabfa85f652f1) with z80 rootfs 0.3 and 0.4pre, with the same result:

[image]

The kernels were built with sdcc from the fuzix sdcc fork.

I have tried the kernel from 2019 built by bruxy.regnet.cz/8bit/fuzix/ - it is affected by #766. It finds the rootfs and init, and starts init, but then goes into loop saying "sd: failed, retrying.". Basically, the reason I've tried [600f2cbc0ecbc01fbb5ca6c825fbabfa85f652f1](https://github.com/EtchedPixels/FUZIX/commit/600f2cbc0ecbc01fbb5ca6c825fbabfa85f652f1)  is because it's around the date when @pawosm-arm mentioned that the OS now boots fine. So, I am wondering if it's something wrong with how I build the kernel... 

Could you please help me to figure out what's happening?

ninakali, 2023-03-30:
FWIW, the partition table I have is
`Device     Boot   Start     End Sectors Size Id Type
/dev/sde1          2048 4196351 4194304   2G  b W95 FAT32
/dev/sde2       4196352 6293503 2097152   1G  b W95 FAT32
/dev/sde3       6293504 6359039   65536  32M 7e unknown
/dev/sde4       6359040 6367231    8192   4M 7f unknown`

I've built af6bd7bd19dcd12206da7bf5070d979c6de8530e with patch from https://github.com/EtchedPixels/FUZIX/issues/766#issuecomment-661948737 - with the same "init not found" error. I seriously suspect my toolchain now :/

ninakali, 2023-03-30:
Hm, the build from 2019 doesn't work with 0.4pre rootfs -- "init not found" yet again. I'm going to try the  newer kernel with 0.3 image once more. Maybe I just messed up something...

ninakali, 2023-03-30:
@pawosm-arm 

No, 3 is the same as hda3: https://github.com/EtchedPixels/FUZIX/blob/70e28d1df6129ca5e676290ce7d9607190aa75ea/Kernel/start.c#L142
Fwiw, I've tried using hda3 with the same result.

If you have FUZIX.BIN from that build laying around, could you please share it with me? Having symbols and maybe object files could help, too, so I could find the difference between my build and yours, in case if it will work.

pawosm-arm, 2023-03-30:
Indeed it's different. And I'm using +3 build as I'm running it on +3e hardware. AFAIK zxdiv is for Spectrum 128 with DivIDE.

ninakali, 2023-03-30:
Ugh, tough pickle. I have +3 with DivMMC which is supposed to be sort of compatible with DivIDE... And it seems to be somewhat close to working, because the kernel finds the partition table just right, and on old kernels even reads init...

pawosm-arm, 2023-03-30:
If the old kernel starts `init`, then it's clearly a regression.

EtchedPixels, 2023-03-30:
It should work. It's clearly getting to the point it starts loading init, which means the low level disk access was working (it read the partition tables, superblock etc already).

When you say you used the 0.4rc1 root fs is that the one you built with the kernel ? as the ABI has changed a bit during that period.

EtchedPixels, 2023-03-30:
Works for me if I use the root fs built from git.

Do need to find a sane way to get "make diskimage" working on a spectrum

EtchedPixels, 2023-03-30:
make diskimage added, that was fun

ninakali, 2023-03-30:
Thank you! It was indeed my fault; rebuilding the latest kernel and the latest rootfs helped. I had to use separate compilers for rootfs (sdcc 4.0) and kernel (3.9 patched) though.

closed 2023-03-30 by ninakali