EtchedPixels/FUZIX #985: rcbus-z180 build errors

electrified, 2023-04-17:
Hi there,

I get some i2c related errors when attempting to build the rcbus-z180 platform.

```
../dev/i2c.c:16: error 101: too many parameters 
../dev/i2c.c:19: error 101: too many parameters 
../dev/i2c.c:21: error 101: too many parameters 
../dev/i2c.c:23: error 101: too many parameters 
../dev/i2c.c:24: error 101: too many parameters 
../dev/i2c.c:30: error 101: too many parameters 
../dev/i2c.c:35: error 101: too many parameters 
../dev/i2c.c:36: error 101: too many parameters 
../dev/i2c.c:37: warning 85: in function i2c_receive unreferenced function argument : 'bus'
../dev/i2c.c:37: warning 85: in function i2c_receive unreferenced function argument : 'dev'
../dev/i2c.c:37: warning 85: in function i2c_receive unreferenced function argument : 'buf'
../dev/i2c.c:44: warning 112: function 'i2c_claim_bus' implicit declaration
../dev/i2c.c:47: warning 112: function 'i2c_start' implicit declarationsdasz80  -o syscall_umask.rel syscall_umask.s

../dev/i2c.c:49: warning 112: function 'i2c_write' implicit declaration
../dev/i2c.c:51: warning 112: function 'i2c_stop' implicit declaration
../dev/i2c.c:52: warning 112: function 'i2c_release_bus' implicit declaration
../dev/i2c.c:58: warning 112: function 'i2c_write' implicit declaration
../dev/i2c.c:63: ../dev/i2c_bitbang.c:12: warning 112: warning 112: function 'i2c_stop' implicit declarationfunction 'i2c_set' implicit declaration

../dev/i2c.c:64: warning 112: function 'i2c_release_bus' implicit declaration
../dev/i2c_bitbang.c:14: warning 112: function 'i2c_scl' implicit declaration
../dev/i2c.c:44: error 101: too many parameters 
../dev/i2c_bitbang.c:12: ../dev/i2c.c:47: error 101: error 101: too many parameters too many parameters 

../dev/i2c_bitbang.c:14: ../dev/i2c.c:49: error 101: error 101: too many parameters too many parameters 

-:0: ../dev/i2c.c:51: warning 85: error 101: in function i2c_clock_up unreferenced function argument : 'bus'too many parameters 

../dev/i2c.c:52: error 101: too many parameters 
../dev/i2c.c:58: error 101: too many parameters 
../dev/i2c.c:63: error 101: too many parameters 
../dev/i2c.c:64: error 101: too many parameters 
../dev/i2c.c:65: warning 85: in function i2c_send unreferenced function argument : 'bus'
../dev/i2c.c:65: warning 85: in function i2c_send unreferenced function argument : 'dev'
../dev/i2c.c:65: warning 85: in function i2c_send unreferenced function argument : 'buf'../dev/i2c_bitbang.c:20: 
warning 112: function 'i2c_set' implicit declaration
../dev/i2c_bitbang.c:21: warning 112: function 'i2c_set' implicit declaration
../dev/i2c_bitbang.c:22: warning 112: function 'i2c_clear' implicit declaration
../dev/i2c_bitbang.c:23: warning 112: function 'i2c_clear' implicit declaration
../dev/i2c_bitbang.c:20: error 101: too many parameters 
../dev/i2c_bitbang.c:21: error 101: too many parameters 
../dev/i2c_bitbang.c:22: error 101: too many parameters 
../dev/i2c_bitbang.c:23: error 101: too many parameters 
../dev/i2c_bitbang.c:23: warning 85: in function i2c_start unreferenced function argument : 'bus'
../dev/i2c_bitbang.c:29: warning 112: function 'i2c_clear' implicit declaration
../dev/i2c_bitbang.c:30: warning 112: function 'i2c_clear' implicit declaration
../dev/i2c_bitbang.c:31: warning 112: function 'i2c_set' implicit declaration
../dev/i2c_bitbang.c:32: warning 112: function 'i2c_set' implicit declaration
../dev/i2c_bitbang.c:29: error 101: too many parameters 
../dev/i2c_bitbang.c:30: error 101: too many parameters 
../dev/i2c_bitbang.c:31: error 101: too many parameters 
../dev/i2c_bitbang.c:32: error 101: too many parameters 
../dev/i2c_bitbang.c:32: warning 85: in function i2c_stop unreferenced function argument : 'bus'
sdasz80  -o syscall__statfs.rel syscall__statfs.s
../dev/i2c_bitbang.c:41: warning 112: function 'i2c_set' implicit declaration
../dev/i2c_bitbang.c:45: warning 112: function 'i2c_sda' implicit declaration
../dev/i2c_bitbang.c:47: warning 112: function 'i2c_clear' implicit declaration
../dev/i2c_bitbang.c:49: warning 112: function 'i2c_clear' implicit declaration
../dev/i2c_bitbang.c:51: warning 112: function 'i2c_clear' implicit declaration
../dev/i2c_bitbang.c:41: error 101: too many parameters 
../dev/i2c_bitbang.c:45: error 101: too many parameters 
../dev/i2c_bitbang.c:47: error 101: too many parameters 
../dev/i2c_bitbang.c:49: error 101: too many parameters 
../dev/i2c_bitbang.c:51: error 101: too many parameters 
sdasz80  -o syscall_execve.rel syscall_execve.s
../dev/i2c_bitbang.c:63: warning 112: function 'i2c_set' implicit declaration
../dev/i2c_bitbang.c:65: warning 112: function 'i2c_clear' implicit declaration
../dev/i2c_bitbang.c:68: warning 112: function 'i2c_clear' implicit declaration
../dev/i2c_bitbang.c:73: warning 112: function 'i2c_set' implicit declaration
../dev/i2c_bitbang.c:75: warning 112: function 'i2c_sda' implicit declaration
../dev/i2c_bitbang.c:77: warning 112: function 'i2c_clear' implicit declaration
../dev/i2c_bitbang.c:63: error 101: too many parameters 
../dev/i2c_bitbang.c:65: error 101: too many parameters 
../dev/i2c_bitbang.c:68: error 101: too many parameters 
../dev/i2c_bitbang.c:73: error 101: too many parameters 
../dev/i2c_bitbang.c:75: error 101: too many parameters 
../dev/i2c_bitbang.c:77: error 101: too many parameters 
../dev/i2c_bitbang.c:92: warning 112: function 'i2c_claim_bus' implicit declaration
make[2]: *** [Makefile:35: i2c.rel] Error 1
../dev/i2c_bitbang.c:100: warning 112: make[2]: *** Waiting for unfinished jobs....
function 'i2c_release_bus' implicit declaration
../dev/i2c_bitbang.c:112: warning 112: function 'i2c_release_bus' implicit declaration
../dev/i2c_bitbang.c:92: error 101: too many parameters 
../dev/i2c_bitbang.c:97: error 20: Undefined identifier 'I2C_READ'
../dev/i2c_bitbang.c:97: error 20: Undefined identifier 'I2C_READ'
../dev/i2c_bitbang.c:100: error 101: too many parameters 
../dev/i2c_bitbang.c:101: error 20: Undefined identifier 'udata'sdasz80  -o syscall__getdirent.rel syscall__getdirent.s

../dev/i2c_bitbang.c:101: error 25: Structure/Union expected left of '.->'
../dev/i2c_bitbang.c:112: error 101: too many parameters 
../dev/i2c_bitbang.c:120: warning 112: function 'i2c_claim_bus' implicit declaration
../dev/i2c_bitbang.c:128: warning 112: function 'i2c_release_bus' implicit declaration
../dev/i2c_bitbang.c:140: warning 112: function 'i2c_release_bus' implicit declaration
../dev/i2c_bitbang.c:120: error 101: too many parameters 
../dev/i2c_bitbang.c:125: error 20: Undefined identifier 'I2C_WRITE'
../dev/i2c_bitbang.c:125: error 20: Undefined identifier 'I2C_WRITE'
../dev/i2c_bitbang.c:128: error 101: too many parameters 
../dev/i2c_bitbang.c:129: error 20: Undefined identifier 'udata'
../dev/i2c_bitbang.c:129: error 25: Structure/Union expected left of '.->'
../dev/i2c_bitbang.c:140: error 101: too many parameters 
sdasz80  -o syscall_setuid.rel syscall_setuid.s
sdasz80  -o syscall_setgid.rel syscall_setgid.s
make[2]: *** [Makefile:35: i2c_bitbang.rel] Error 1
make[2]: Leaving directory '/home/ed/dev/3rdparty/FUZIX/Kernel/platform-rcbus-z180'
sdasz80  -o syscall__time.rel syscall__time.s
make[1]: *** [Makefile:115: target] Error 2
make[1]: Leaving directory '/home/ed/dev/3rdparty/FUZIX/Kernel'
sdasz80  -o syscall__stime.rel syscall__stime.s
sdasz80  -o syscall_ioctl.rel syscall_ioctl.s
sdasz80  -o syscall_brk.rel syscall_brk.s
make: *** [Makefile:142: kernel] Error 2
make: *** Waiting for unfinished jobs....
```

EtchedPixels, 2023-04-17:
Yep I've been redesigning the i2c code and not got to fixing up the bitbang stuff yet

electrified, 2023-04-17:
Thanks. I've rolled back to https://github.com/EtchedPixels/FUZIX/commit/279542b720cc7438b89a9d6c9c60830f63182db4 prior to the i2c changes which builds.

closed 2023-04-18 by EtchedPixels