EtchedPixels/FUZIX #987: Port lrzsz for Fuzix for transferring files with computers

jwy2008, 2023-04-21:
When I was using Fuzix built for Raspberry Pi Pico, I felt the need to have a program that could transfer files to and from the computer, such as lrzsz, many Raspberry Pi pico projects have ported it, such as Luapico and pshell, I tried to port it myself, but because I didn't know much about it, it failed, so I wanted to be able to build an lrzsz for Fuzix (and because I don't speak English very well, So the above content is from bing translation, so the readability may be poor,   please forgive me)

EtchedPixels, 2023-04-21:
ucp will let you directly modify the Fuzix file system. An xmodem would be useful but lrzsz is horribly messy and too large for some of the smaller systems Fuzix supports

jwy2008, 2023-04-21:
> ucp will let you directly modify the Fuzix file system. An xmodem would be useful but lrzsz is horribly messy and too large for some of the smaller systems Fuzix supports

I guess I should have confused lrzsz with xmodem, and thanks for the reply!

closed 2023-05-22 by EtchedPixels