EtchedPixels/FUZIX #1008: Running cromemco port

ajcasado, 2023-06-27:
Hi, I'm trying to run the cromemco port but it fails to boot:
```
Standby
FUZIX LOADER
..................
FUZIX version 0.4pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2023 Alan Cox <alan@etchedpixels.co.uk>
Devboot
448kB total RAM, 384kB available to processes (6 processes max)
Enabling interrupts ... ok.
bootdev: fd1
Mounting root fs (root_dev=257, ro): fd1: error 0010
failed
```
I guess that my problem is that as is said in the cromemco platform README file, the cromemco emulator has to be patched to "understand Fuzix DS/DD media".
Where can I get this patch for the cromemco emulator?

EtchedPixels, 2023-06-29:
Does look that way. I sent it upstream long ago so I thought it had been applied there.

Added it to Kernel/patches and pushed

ajcasado, 2023-07-03:
Thanks!
I'll try to test it this week. I'm interested on this port as I guess that the banked memory model could be used with the Amstrad CPC standard expansion ram using the C2 memory mode.

EtchedPixels, 2023-07-03:
That was one of the reasons I never tackled the CPC. The base 128K machine isn't too hard as it looks like you can flip between 0/1/2/7 and 4/5/6/7 and put the video in 3 much like the ZX Spectrum +3 (unsurprisingly given the origin of the +3) but the extra memory looked difficult

ajcasado, 2023-07-04:
The patch was already applied in the cromemco emulator, in fact the comments at the header of the file cromemco-fdc.c says that it is applied since SEP-2019, so that was not the problem.
Any clue about the meaning of "error 0010" in: `Mounting root fs (root_dev=257, ro): fd1: error 0010` ?
About the Amstrad CPC, my idea is to switch between 64K pages as the cromemco do. The second 64K page will be used for the kernel space, from the third to the last for user space (standard expansions can go up to 2MB, usually 512k and 1MB). The first will be used for booting, for the video RAM (that can be placed anywhere in this page and can use 16K and 32K modes) and also to hold the CPC firmware RAM so we can use firmware calls to manage the hardware (the firmware implements a VT52 terminal and manages the standard ROM system). Using two 16K banks for VRAM two tty consoles can be implemented, with one 32K bank we can use overscan mode (no border). There are also expansion devices as the M4 board that provides wifi and sd storage, the USIFAC that provides USB storage and serial port or the albireo that provides a sd card and an USB port that can be managed from the first page calling already existing ROM routines.

EtchedPixels, 2023-07-05:
Humm... looking at the other changes and the later FDC controller.

Does editing cromemco.s and changing 

````
        in      a,(0x30)
        and     #0xFC                   ; Mask off bad bits
jrreta: jr      reta                    ; HL return
````

in the fd_done routine to and #0xEC make it work ?

EtchedPixels, 2023-07-05:
(oh and maybe alsoi the and #0x98 at the end of _fd_seek)

ajcasado, 2023-07-06:
> Humm... looking at the other changes and the later FDC controller.
>
> Does editing cromemco.s and changing
>
> ```
>         in      a,(0x30)
>         and     #0xFC                   ; Mask off bad bits [...]

No luck, this doesn't fix the issue.

> (oh and maybe alsoi the and #0x98 at the end of _fd_seek)

What value is should be the correct instead 0x98?

EtchedPixels, 2023-07-06:
Would be 0x88.

Guess I need to grab a current version of the emulation and see what is up (I am suspecting it's the emulator is now emulating the later floppy controller and correctly which probably means I have some bit masks wrong)

closed 2024-06-16 by EtchedPixels