EtchedPixels/FUZIX #1015: Building RCBus-Z180 (SC126) toolchain and FUZIX

Arawn-Davies, 2023-08-13:
Hi, I'd like to try building a more updated version of FUZIX for my SC126 (or my SC130), but I have no clue where to start.

I've read I need to build the downstream fork of SDCC (https://github.com/EtchedPixels/sdcc280) which fixes a few bugs which are present upstream, but I've no idea where to start.as the wiki and other materials aren't fully clear to me

I've built i686-elf binutils/GCC/GDB (for x86 OS development) but as I'm getting into Z(x)80 development, I'm willing to give it a shot if someone could please provide a how-to :)  Thanks!

EtchedPixels, 2023-08-14:
You should be able to git clone the sdcc280 tree and then build it and install it exactly as per the SDCC instructions. It's a fork of sdcc 3.8 era sdcc with additions for banking and some fixes. SDCC 4 changed API but for the SC126 the older SDCC should also work (3.8.x where x is whatever was last) as the SC126 doesn't use the banking extensions.

In theory then it's as simple as going into the Fuzix tree, setting the target in Makefile and typing "make diskimage".

Arawn-Davies, 2023-08-15:
I tihnk it might be worth putting on the wiki that libboost and zlib (maybe one or two others) are needed to compile, but tried various fresh installs of linux in a vm with gcc etc. but can't find a good working version combo yet

Arawn-Davies, 2023-08-18:
I had to build latest versions of libboost and gputil to successfully compile SDCC and Fuzix on Ubuntu (both WSL and 23.04 in VM) , but fully works now :)

closed 2023-08-18 by Arawn-Davies