EtchedPixels/FUZIX #1017: no tx on UART on 8266

qqmajikpp, 2023-08-20:
I got the 8266 flashed and the sd card imaged. I can connect via uart and see output but when it asks me to input the time, it doesnt seem to accept. :(

EtchedPixels, 2023-08-21:
Does it echo the characters and not accept the newline or do anything or does it not even echo ?

qqmajikpp, 2023-08-21:
it does appear to let me type, on that note, the cursor moves freely and i
can backspace all the way to the left past the prompt. i can type but
backspace doesn't actually seem to delete anything nor does any entry seem
to register.

~Ekadasi Newton (a.k.a. Scotty)

Wanna know how to pronounce Ēkādasī? Click this link and then click the
Listen (speaker) icon in the lower right hand corner of the right box.
<https://translate.google.com/#en/hi/ekadasi>

I also have a profile HERE <http://www.mormon.org/me/f0pn>

EtchedPixels, 2023-08-21:
Unable to reproduce this can you post the full boot messages and compiler version used ?

qqmajikpp, 2023-08-21:
thinking about it now, i couldnt actually get Index of
/downloads/0.4/esp8266 (fuzix.org) <http://fuzix.org/downloads/0.4/esp8266/> to
boot for me. i'll have to post the logs when i get a chance. im actually
using http://cowlark.com/2021-02-09-esp8266-fuzix/fuzix.tar.xz so thats
probably why...

~Ekadasi Newton (a.k.a. Scotty)

Wanna know how to pronounce Ēkādasī? Click this link and then click the
Listen (speaker) icon in the lower right hand corner of the right box.
<https://translate.google.com/#en/hi/ekadasi>

I also have a profile HERE <http://www.mormon.org/me/f0pn>

EtchedPixels, 2023-08-21:
If you can reproduce it in 0.4 then I'm interested, if it's just the old release then no.

You'll need to build the flash images as well as they are not in the Image set (for one you need to adjust config.h for your setup and whether you have  1 SD, 2 SD or SD and Wiznet 5500 as well as setting the root device to 0 or 0x11 depending upon if you are booting from flash or SD

EtchedPixels, 2023-08-24:
Not upstream tree, not reproducable, no response

closed 2023-08-24 by EtchedPixels