EtchedPixels/FUZIX #1018: RC2014 w/ TMS9918A: weird behavior with multiple TTYs

ry755, 2023-08-26:
Switching to a different TTY via the alt-Fx keys works, but upon pressing any key within that different TTY the computer hangs and the displayed cursor starts blinking quickly as if characters were being sent very quickly, but nothing appears on screen.

Also, spawning a shell on an alternate TTY (via editing /etc/inittab) partially works, it shows the login prompt but again, same issue with attempting to enter characters. This also applies to spawning a shell on the serial port (/dev/tty5), no characters can be entered.

This can be reproduced in the emulator with the following command: `./rc2014_sdl2 -s -r RCZ80_std.rom -b -i emu-ide.img -P -T`

EtchedPixels, 2023-08-27:
Interesting - I've seen this on the emulation before it but it disappeared when I tried to debug it. I will have another look into it when I get time

EtchedPixels, 2023-10-04:
Fixed. Silly bug with queue setup

closed 2023-10-04 by EtchedPixels