EtchedPixels/FUZIX #1019: Somehow fit w5x00 on msx2

retrodeluxe, 2023-08-29:
I have been trying to fit networking and the w5x00 driver on an msx2 kernel and just realized is not going to work because after removing some necessary stuff I am still about 7kb short to get everything not common under 0xC000.

Do you have any suggestions on how to approach this?

I could try to extend the kernel to use one or two more 16Kb pages and fit some drivers there. The page swap can be handled by the sdcc banking support; I have some sdcc patches I could try to backport to the sdcc280 used with Fuzix.

retrodeluxe, 2023-08-29:
oh wait, I just found the external-banker patch and the bihx tool. I can replicate the same as in other platforms.

EtchedPixels, 2023-08-31:
The best example for that is probably the rc2014. That has the common in the top 16K, the bottom 16K data/bss and the middle 32K is paged.

closed 2024-06-16 by EtchedPixels