EtchedPixels/FUZIX #1020: RC2014 w/ TMS9918A: VDPIOC_WRITE ioctl fails

ry755, 2023-08-31:
I'm attempting to make an application that utilizes the bitmap (Graphics II) mode. As an initial test, I started by modifying gfxtest.c to test the TMS9918A with the following code:
```c
void test_vdp_9918a(void)
{
    struct vdp_rw vdp;
    uint8_t regs[8] = { 0x02, 0xC0, 0x08, 0xFF, 0x03, 0x76, 0x03, 0x76 };
    uint8_t buf;

    if (ioctl(0, GFXIOC_MAP, &v)) {
        perror("GFXIOC_MAP");
        exit(1);
    }

    if (ioctl(0, VDPIOC_SETUP, &regs[0])) {
        perror("VDPIOC_SETUP");
        ioctl(0, GFXIOC_UNMAP, &v);
        exit(1);
    }

    // attempt to write a byte to the screen table
    buf = 0x00;
    vdp.data = &buf;
    vdp.lines = 1;
    vdp.cols = 1;
    vdp.vdpaddr = 0x1800;
    vdp.stride = 0;
    if (ioctl(0, VDPIOC_WRITE, &vdp)) {
        perror("VDPIOC_WRITE");
        ioctl(0, GFXIOC_UNMAP, &v);
        exit(1);
    }
}
```

Everything works up until the VDPIOC_WRITE ioctl, which returns EFAULT/"Bad address". I have tracked it down to [this block of code](https://github.com/EtchedPixels/FUZIX/blob/master/Kernel/platform-rc2014/video-tms9918a.c#L367-L370). I can't tell if this is a bug in FUZIX (not likely, considering every TMS9918A-supporting platform I've looked at uses the same ioctl handling code) or, more likely, I'm just doing something wrong.

The docs/Graphics.md document mentions that these ioctls are not yet finished. Is this already a known issue? Or am I just doing something wrong here?

EtchedPixels, 2023-09-02:
The read/write ones are not tested yet. The older code used the map/setup then hit the I/O ports given by the map

ry755, 2023-09-03:
Unfortunately that didn't fix the issue described here, still getting "Bad address".

EtchedPixels, 2023-10-04:
Should also be fixed this time around 8)

closed 2024-01-21 by EtchedPixels