EtchedPixels/FUZIX #1021: Plus +3 wiznet

EtchedPixels, 2023-09-04:
Plus 3 kernel squashed, now seems to fit networking and more. Need to write the low level wiznet glue including the crazy mode
bank flipping sequences.

@pawosm-arm FYI

pawosm-arm, 2023-09-04:
Hi Alan,

That's a great news! I'm experiencing extremely busy weeks recently, but I'll try to give it a go when the tide gets low again.

I'm slightly confused that the change was introduced on the `zx+3` platform, while `zx-spectra` hasn't been touched.

Thanks,
Paul

EtchedPixels, 2023-09-04:
I managed to squash the +3 kernel down enough to do an unbanked solution. Will look at spectra for straight 128K later

pawosm-arm, 2023-09-04:
Ok then, it means `zx+3` is what I am after, and `zx-spectra` is just a plan-b should the `crazy mode bank flipping sequences` turn out to be too problematic.

pawosm-arm, 2023-09-06:
I think `plt_ide.h` needs to be copied over from the `platform-zxdiv`, otherwise `tinyide.c` won't compile.

pawosm-arm, 2023-09-14:
Strange things are happening when I'm trying to build the `zx+3` filesystem image:

```
ERROR stdint.h:34 file not found
ERROR dbg.h:73 file not found
ERROR dio.h:74 file not found
ERROR device.h:75 file not found
ERROR serial.h:76 file not found
ERROR conio.h:77 file not found
ERROR drivewire.h:125 file not found
ERROR userstructs.h:143 file not found
ERROR stdint.h:168 file not found
ERROR Fix errors and re-run
```

EtchedPixels, 2023-09-15:
Should also be fixed. Things broke a bit from adding the 8080/5 native compiler set up. The Z80 one needs thought as the native and cross compilers are different so libc needs building twice somehow

pawosm-arm, 2023-10-21:
looking at the function stubs in the `main.c` file (why not `wiznet.c`?) I can't figure out how it's going to model `CPLD`.

EtchedPixels, 2023-10-21:
Not sure I understand the question. The stubs are expecting to be able to drive the Wiznet chip (presumably by switching all the memory maps back and forth)

pawosm-arm, 2023-10-26:
Well, the only WizNet device I have is the Spectranet. I could be wrong and maybe I'm seeing non-existing problem, but for me, it doesn't seem doable to talk to the chip directly bypassing CPLD. So unless those once mentioned changes in the Spectranet firmware are about facilitating that, or we're talking about some abstract WizNet device and Spectranet isn't just one of them, and I should rather look at the `zxspectra` platform...

EtchedPixels, 2023-10-26:
The CPLD doesn't appear to do anything magic. It's just mapping the device directly into the Z80 address space. The device has 3 modes

- SPI mode
- A mode where it occupies only a few addresses and you poke at it by setting index registers
- A mode where it's properly mapped

The wiznet using systems currently in the kernel use the  few addresses mode and indexes. The CPLD merely maps the device 128K into a load of 4K pages in the spectranet space when paged in. Either way of accessing them should work although I imagine mapping it and writing directly will be faster than pinning the one page and using the indirect modes