EtchedPixels/FUZIX #1028: Dragon 64 Serial terminal sends but doesn't receive

tjewell99, 2023-09-23:
(it's possible this isn't an 'issue' and is me not having this set up correctly)

I'm trying to get a dumb terminal working with the serial port on my Dragon 64. I change inittab as follows to pick up the serial port on tty5:

# Terminals
01:3:respawn:getty /dev/tty1
02:3:respawn:getty /dev/tty5

Once booted a login prompt appears just fine - but doesn't respond to any input. I can't login, nothing happens. I'm 99% sure the terminal and serial setup is fine, as it all behaves when I use a terminal program under Flex on the Dragon. Have I possibly missed a step when I set this up?

Additional question - how do I change the baud rate of the getty? I've got a modem I'd like to hookup, and that wants to run at 1200 baud.

Many thanks in advance!

EtchedPixels, 2023-09-23:
I've not tested the Dragon 64 serial except for some minimal emulator poking so it is probably broken. I will take a look

For the baud you can specify more on the getty line

eg

02:3:respawn:getty /dev/tty5 1200 vt100 80 25

for a vt100 at 80x25 at 1200 baud

tjewell99, 2023-09-23:
Interestingly, I note that if do:

02:3:respawn:getty /dev/tty5 1200

The baudrate is actually 2400.

EtchedPixels, 2023-09-23:
On no surprise - means the table is wrong. I imagine they are all out by x2 ?

I've pushed a change for the port I hope. I'd never wired up th receive interrupt side as I needed to figure out how to properly detect it so it didn't go off doing crazy things on other machines.

EtchedPixels, 2023-09-23:
Speed one is a silly bug so should also now be fixed

tjewell99, 2023-09-24:
Hi! Speed bug is fixed (thanks!), but I still don't get any input from the terminal, so not sure if interrupt is working?

tjewell99, 2023-09-24:
A little bit of progress ... I'm using a mooh, so I added tty_detect to Kernel/platform-dragon-mooh/discard.c too. However, tty_detect doesn't seem to actually detect the ACIA. If I force has_acia to 1 regardless, I can see tty_interrupt being called - and to these eyes, the check for the RX data register on the ACIA status (0x08 on FF05) looks okay. But it's not seeing anything and hence it's not reading the terminal, and I can't (yet) see why.

tjewell99, 2023-09-26:
Still pondering this one, a couple things ...

Kernel/platform-dragon-nx32/devtty.c

Should lines 638 and 641 read 
*uart_command = 0x03;
rather than
uart_command = 0x03;

tty_setup in the same file, should there be a
*uart_command = r;
after line 194, otherwise the value doesn't seem to get set?  Also, I cannot find for life of me where tty_setup is called from.

And I don't really understand what tty_detect is doing? If I'm reading my 6551 datasheet correctly, it's checking for a framing error (&2), doing a 'reset' (enabling DTR? 0x03) and then checking for the error again. It then sets up the ACIA in a way I don't understand - 0x96 suggests an echo on, a transmit interrupt and disable DTR again.  But I could be soooo reading this wrongly :)

However, setting the ACIA to my usual default of 0x0A seems to make no difference, nor does making the other changes I mentioned above, so it's quite possible I am barking up absolutely the wrong tree!

tjewell99, 2023-09-28:
Hi! The 6551 detect is now working, but it's still not responding to any input (although the login prompt comes up just fine). 

I don't know if this will help at all, but this is what the ROM routine does when reading data (although of course, it locks the machine completely while reading it, which is less than ideal!).

I'm just wondering if there's a clue with what it's doing to DTR - which it seems to enable before reading and disable after reading.

```
        ldb $ff06
        orb #$21    ; No parity, DTR low
        stb $ff06
        andb    #$fe
        lda #$08    ; RX data reg full
rxwait      bita    $ff05
        beq rxwait  ; Wait until we've got something to read
        stb $ff06
        lda $ff04
```