EtchedPixels/FUZIX #1032: when compiling platform rpipico, pico is not the default...

janneman001, 2023-10-19:
When compiling platform rpipico it is not the default.

When i define CONFIG_RC2040 in config.h it becomes the default and i can use the sdcard pins from rpi pico board again according to wiring.jpg in doc directory of the platform.

EtchedPixels, 2023-10-19:
Humm this does look backwards unless I am missing something @davidgiven ?

(it's not a platform I know well - its about 30 years too modern)

davidgiven, 2023-10-19:
It's late, but it looks like _without_ CONFIG_RC2040 it uses that standard rpi pins --- 2, 3, 4, 5. See https://github.com/EtchedPixels/FUZIX/blob/v0.4/Kernel/platform-rpipico/devsdspi.c#L14. So I'm not sure what you're seeing.

You might want to try putting a `#error` in `devsdspi.c` to make sure that the correct preprocessor branch is taken.

dennowiggle, 2024-03-07:
> It's late, but it looks like _without_ CONFIG_RC2040 it uses that standard rpi pins --- 2, 3, 4, 5. See https://github.com/EtchedPixels/FUZIX/blob/v0.4/Kernel/platform-rpipico/devsdspi.c#L14. So I'm not sure what you're seeing.

That is correct. Without adding "CONFIG_RC2040" in config,h  it uses the standard rpi pins --- 2, 3, 4, 5.  This is the default due to config.h containing "#undef CONFIG_RC2040". This does not match the wiring diagram of wiring.jpg in the doc directory of the platform which shows 12,13,14,15. 

12,13,14,15 is used in the Raspberry Pi website instructions for running fuzix on Pi Pico.
https://www.raspberrypi.com/news/how-to-get-started-with-fuzix-on-raspberry-pi-pico/

In "devsdspi.c" line 28 shows a comment "David Given's Arrangement", but if the raspberry pi website arrangement and the wiringpi.jpg image both reference "Pico SPI GPIO connected to SD SPI1" which seems to be the real "David Given's Arrangement". I think that is where the confusion is.

With current config.h SD SPI0 pins 2345 are used. If I wire per wiring.jpg I get "no card found". If I wire per 2345 I get a hang. So I'm stuck here now. This is the hang (using EtchedPixels repository).
```
FUZIX version 0.5
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2024 Alan Cox <alan@etchedpixels.co.uk>
Devboot
264KiB total RAM, 160KiB available to processes (15 processes max)
Enabling interrupts ... ok.
NAND flash, 1952kB physical 1277kB logical at 0x13018000: hda:
SD drive 0:
```
Is Fuzix 5.0 working with rpipico and the latest Pico SDK?

I did try to build with David Givens repository following the Raspberry Pi website instructions but had the same result.

I burned the card image onto sda2 following the instructions on the Raspberry Pi website.

Could the new issue of non booting SD card be due to the error I get when running "./update-flash.sh"
```
Making 512 byte/block filesystem with normal byte order on device filesystem.img with fsize = 2547 and isize = 32.
Source file '../../../Standalone/filesystem-src/etc-files/issue' not found
chmod: can't stat file 'issue': 2
```

veremenko-y, 2024-06-26:
The mismatch between configuration and wiring.jpg has been fixed in #1074 

>Is Fuzix 0.5 working with rpipico and the latest Pico SDK?
@dennowiggle 
Running just fine. There's also `make image-sd` now that makes 32MB file system for SD card.
I've noticed sometimes dd did not actually copied the image if I tried to dd into the partition. It did work if I dd into the device and specify the offset. E.g. `dd if=filesystem.img of=/dev/sdX bs=512 seek=<partition start>`