EtchedPixels/FUZIX #1040: FUZIX for Mizar32: Thoughts and hacks so far

ramangopalan, 2024-01-17:
Dear Alan, Greetings!

Firstly, many thanks for FUZIX! I had a lot of fun reading the
codebase. @sergio-sorrenti and I were talking about exploring FUZIX
for the Mizar32 [1] platform a few weeks ago. I thought I will give it
a shot.

The first problem I had with AT32UC3 controllers (also in the context
of FUZIX) was the lack of debug capability on GNU/Linux. I invoked GNU
Make on the rpi-pico platform to figure out the core kernel files that
were being compiled. I imported all of them into a test Microchip
Studio project for Mizar32-A (AT32UC3A0512) on Windows.

I was able to get kprintf to work very quickly. After that, I
struggled with the SD card interface on Mizar32 for a bit. I
implemented __hard_irqrestore, __hard_ei, __hard_di and everything
else the compiler asked for on Windows.

[image]

I didn't have the root file system yet. But I fired up the debug
session with rpi-pico's root file system anyway. At least it came to a
point where it complained about the endianness. This part was too
exciting: just proof that the bare metal for the SD card and kprintf
(USART) works.

[image]

I knew that init wasn't going to fire. I assumed that context
switching OR interrupts weren't important for my debug sessions at
that moment. I took it as it came. AVR32 is a big endian machine.
s_mounted on the debug session held 50737. Perfect.

[image]

I knew it was time to checkout the root file system for AVR32. I went
back to GNU/Linux, cloned FUZIX and took much inspiration from 68K,
the ARM M0 and M4 ports.

I then spent time hacking Makefiles and writing files for dependencies
within Library/libs and Library/tools; For root. I implemented a very
quick and (for sure) buggy setjmp. I am sure my longjmp will fail with
what I've written so far. I thought I can fix them later.

I am stuck at `-pie' not being recognized by avr32-gcc.

```
avr32-gcc -mpart=uc3a0512 -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -g -Os -c -I../include  curses/endwin.c -o curses/endwin.o
avr32-gcc -mpart=uc3a0512 -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -g -Os -c -I../include  curses/wgetch.c -o curses/wgetch.o
avr32-gcc -mpart=uc3a0512 -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -g -Os -c -I../include  curses/mvcursor.c -o curses/mvcursor.o
avr32-gcc -mpart=uc3a0512 -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -g -Os -c -I../include  curses/curs_set.c -o curses/curs_set.o
avr32-gcc -mpart=uc3a0512 -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -g -Os -c -I../include  curses/options.c -o curses/options.o
avr32-gcc -mpart=uc3a0512 -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -g -Os -c -I../include  curses/winmove.c -o curses/winmove.o
avr32-ar rc cursesuc3a0512.lib curses/scan.o curses/setterm.o curses/initscr.o curses/wclrtobot.o curses/cursesio.o curses/werase.o curses/wbox.o curses/tabsize.o curses/windel.o curses/wintouch.o curses/overlay.o curses/waddstr.o curses/move.o curses/newwin.o curses/wclrtoeol.o curses/refresh.o curses/wdelch.o curses/winsertln.o curses/unctrl.o curses/winscrol.o curses/winsch.o curses/prntscan.o curses/charpick.o curses/prnt.o curses/waddch.o curses/scrreg.o curses/wclear.o curses/wgetstr.o curses/doupdt.o curses/wdeleteln.o curses/endwin.o curses/wgetch.o curses/mvcursor.o curses/curs_set.o curses/options.o curses/winmove.o
ln -sf cursesuc3a0512.lib libcursesuc3a0512.a
avr32-ar rc readlineuc3a0512.lib readline.o readline_fn.o
ln -sf readlineuc3a0512.lib libreadlineuc3a0512.a
avr32-gcc -mpart=uc3a0512 -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -g -Os -c -I../include  termcap.c
avr32-gcc -mpart=uc3a0512 -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -g -Os -c -I../include  tgetent.c
avr32-ar rc termcapuc3a0512.lib termcap.o tgetent.o
ln -sf termcapuc3a0512.lib libtermcapuc3a0512.a
avr32-ar rc muc3a0512.lib vfscanf_m.o vfprintf_m.o acosf.o acoshf.o asinf.o asinhf.o atan2f.o atanf.o atanhf.o cbrtf.o ceilf.o copysignf.o erff.o expf.o expm1f.o fabsf.o fdimf.o floorf.o fmaxf.o fminf.o fmodf.o frexpf.o hypotf.o ilogbf.o j0f.o j1f.o jnf.o ldexpf.o lgammaf.o lgammaf_r.o logf.o log2f.o log10f.o logbf.o lrintf.o lroundf.o modff.o nearbyintf.o nextafterf.o powf.o remainderf.o remquof.o rintf.o roundf.o scalbnf.o scalbln.o sinf.o sincosf.o sinhf.o sqrtf.o tgammaf.o __expo2f.o __float_bits.o __fpclassifyf.o __log1pf.o __signgam.o mathhelper.o remainder.o acos.o acosh.o asin.o asinh.o atan2.o atan.o atanh.o cbrt.o ceil.o copysign.o erf.o exp.o expm1.o fabs.o fdim.o floor.o fmax.o fmin.o fmod.o frexp.o hypot.o j0.o j1.o jn.o ldexp.o lgamma.o lgamma_r.o log.o log2.o log10.o lrint.o lround.o modf.o nearbyint.o pow.o remquo.o rint.o round.o scalbn.o sin.o sincos.o sinh.o sqrt.o tgamma.o __expo2.o __double_bits.o __fpclassify.o __log1p.o __sin.o __cos.o __tan.o __sindf.o __cosdf.o __tandf.o
ln -sf muc3a0512.lib libmuc3a0512.a
make[1]: Leaving directory '/home/raman/Skullets/fuzix/FUZIX/Library/libs'
make[1]: Entering directory '/home/raman/Skullets/fuzix/FUZIX/Library/libs'
make[1]: 'install' is up to date.
make[1]: Leaving directory '/home/raman/Skullets/fuzix/FUZIX/Library/libs'
(cd Applications; make)
make[1]: Entering directory '/home/raman/Skullets/fuzix/FUZIX/Applications'
(cd util; make -f Makefile.uc3a0512)
make[2]: Entering directory '/home/raman/Skullets/fuzix/FUZIX/Applications/util'
avr32-ld /Library/libs/crt0_uc3a0512.o banner.o -o banner -L/Library/libs -lcuc3a0512 -pie -static -no-dynamic-linker -z max-page-size=4 -L/home/raman/tools/avr32-gnu-toolchain-linux_x86_64/bin/../lib/gcc/avr32/4.4.7/ucr2/ -lgcc -T /Library/elfexe32.ld --no-export-dynamic -Bstatic -no-dynamic-linker
avr32-ld: -pie not supported
make[2]: *** [Makefile.common:130: banner] Error 1
make[2]: Leaving directory '/home/raman/Skullets/fuzix/FUZIX/Applications/util'
make[1]: *** [Makefile:11: util] Error 2
make[1]: Leaving directory '/home/raman/Skullets/fuzix/FUZIX/Applications'
make: *** [Makefile:139: apps] Error 2
raman@foo:~/Skullets/fuzix/FUZIX$
```

I use this version of avr32-gcc (the default that comes with Microchip
Studio) OR the one I've been using for years to compile eLua [2] OR
since recently, RT-Thread [3] for Mizar32 on GNU/Linux:

```
avr32-gcc (AVR_32_bit_GNU_Toolchain_3.4.2_435) 4.4.7
Copyright (C) 2010 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

The biggest problem is that we can't have an updated toolchain for
AVR32.

I understand the problem with -pie. Is there a way to further hacks on
FUZIX for AVR32 with this toolchain? Have I missed anything OR have I
thought of something incorrectly?

Your suggestions will be very helpful. Many thanks once again for
FUZIX!

Highest regards and respect,

References:
[1]: https://en.wikibooks.org/wiki/Mizar32
[2]: https://github.com/elua/elua/tree/master/src/platform/avr32
[3]: https://github.com/RT-Thread/rt-thread/tree/master/bsp/avr32/at32uc3a0256

EtchedPixels, 2024-01-20:
There are two tricks I use depending on the port

1. Generate the binary with all the relocation data preserved, parse it and write relocation information into the a.out binary
2. Where the tool chain is a pain in the backside or just broken I link it at two suitable addresses and then scan the the binaries for changes to generate the relocations. See for example
 riscv32toaout.c

EtchedPixels, 2024-01-20:
You should get to starting the shell before setjmp/longjmp burns you. The Bourne Shell uses it at startup and then for various kind of error and end of job handling

ramangopalan, 2024-01-23:
Dear Alan,

Many thanks for your message. I revisited RISC-V and ESP8266 parts of
the code and went a bit further with your suggestion. 

The build system is able to generate libmuc3a0512.a and further visits
FUZIX/Applications/util. I checked Makefile.esp8266 and touched
something similar (for apps used by esp8266) for AVR32 - minus
`fforth' for now.

```
avr32-gcc -mpart=uc3a0512 -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -g -Os -c -I../include  tgetent.c
avr32-ar rc termcapuc3a0512.lib termcap.o tgetent.o
ln -sf termcapuc3a0512.lib libtermcapuc3a0512.a
avr32-ar rc muc3a0512.lib vfscanf_m.o vfprintf_m.o acosf.o acoshf.o asinf.o asinhf.o atan2f.o atanf.o atanhf.o cbrtf.o ceilf.o copysignf.o erff.o expf.o expm1f.o fabsf.o fdimf.o floorf.o fmaxf.o fminf.o fmodf.o frexpf.o hypotf.o ilogbf.o j0f.o j1f.o jnf.o ldexpf.o lgammaf.o lgammaf_r.o logf.o log2f.o log10f.o logbf.o lrintf.o lroundf.o modff.o nearbyintf.o nextafterf.o powf.o remainderf.o remquof.o rintf.o roundf.o scalbnf.o scalbln.o sinf.o sincosf.o sinhf.o sqrtf.o tgammaf.o __expo2f.o __float_bits.o __fpclassifyf.o __log1pf.o __signgam.o mathhelper.o remainder.o acos.o acosh.o asin.o asinh.o atan2.o atan.o atanh.o cbrt.o ceil.o copysign.o erf.o exp.o expm1.o fabs.o fdim.o floor.o fmax.o fmin.o fmod.o frexp.o hypot.o j0.o j1.o jn.o ldexp.o lgamma.o lgamma_r.o log.o log2.o log10.o lrint.o lround.o modf.o nearbyint.o pow.o remquo.o rint.o round.o scalbn.o sin.o sincos.o sinh.o sqrt.o tgamma.o __expo2.o __double_bits.o __fpclassify.o __log1p.o __sin.o __cos.o __tan.o __sindf.o __cosdf.o __tandf.o
ln -sf muc3a0512.lib libmuc3a0512.a
make[1]: Leaving directory '/home/raman/Skullets/fuzix/FUZIX/Library/libs'
make[1]: Entering directory '/home/raman/Skullets/fuzix/FUZIX/Library/libs'
make[1]: 'install' is up to date.
make[1]: Leaving directory '/home/raman/Skullets/fuzix/FUZIX/Library/libs'
(cd Applications; make)
make[1]: Entering directory '/home/raman/Skullets/fuzix/FUZIX/Applications'
(cd util; make -f Makefile.uc3a0512)
```
Files like bd.c and chmem.c are getting compiled. But some of them are
not. For example, cal.c, who.c and vile.c. Not sure why. Their object
files are produced though.

```
avr32-gcc -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -Os -isystem /home/raman/Skullets/fuzix/FUZIX/Library/include -isystem /home/raman/Skullets/fuzix/FUZIX/Library/include/uc3a0512 -Wno-char-subscripts   -c -o banner.o banner.c
/home/raman/Skullets/fuzix/FUZIX/Library/link/lduc3a0512 /home/raman/Skullets/fuzix/FUZIX/Library/libs/crt0_uc3a0512.o banner.o -o banner -L/home/raman/Skullets/fuzix/FUZIX/Library/libs --gc-sections --emit-relocs -lcuc3a0512 -L/home/raman/tools/avr32-gnu-toolchain-linux_x86/bin/../lib/gcc/avr32/4.4.7/ -lgcc -T /home/raman/Skullets/fuzix/FUZIX/Library/elf2flt.ld 
avr32-gcc -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -Os -isystem /home/raman/Skullets/fuzix/FUZIX/Library/include -isystem /home/raman/Skullets/fuzix/FUZIX/Library/include/uc3a0512 -Wno-char-subscripts   -c -o bd.o bd.c
/home/raman/Skullets/fuzix/FUZIX/Library/link/lduc3a0512 /home/raman/Skullets/fuzix/FUZIX/Library/libs/crt0_uc3a0512.o bd.o -o bd -L/home/raman/Skullets/fuzix/FUZIX/Library/libs --gc-sections --emit-relocs -lcuc3a0512 -L/home/raman/tools/avr32-gnu-toolchain-linux_x86/bin/../lib/gcc/avr32/4.4.7/ -lgcc -T /home/raman/Skullets/fuzix/FUZIX/Library/elf2flt.ld 
avr32-gcc -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -Os -isystem /home/raman/Skullets/fuzix/FUZIX/Library/include -isystem /home/raman/Skullets/fuzix/FUZIX/Library/include/uc3a0512 -Wno-char-subscripts   -c -o cal.o cal.c
cal.c: In function 'main':
cal.c:144: warning: 'year' may be used uninitialized in this function
/home/raman/Skullets/fuzix/FUZIX/Library/link/lduc3a0512 /home/raman/Skullets/fuzix/FUZIX/Library/libs/crt0_uc3a0512.o cal.o -o cal -L/home/raman/Skullets/fuzix/FUZIX/Library/libs --gc-sections --emit-relocs -lcuc3a0512 -L/home/raman/tools/avr32-gnu-toolchain-linux_x86/bin/../lib/gcc/avr32/4.4.7/ -lgcc -T /home/raman/Skullets/fuzix/FUZIX/Library/elf2flt.ld 
avr32-ld: BFD (AVR_32_bit_GNU_Toolchain_3.4.2_435) 2.23.1 assertion fail /data2/home/toolsbuild/jenkins-knuth/workspace/avr32-gnu-toolchain/src/binutils/bfd/elflink.c:8029
avr32-objcopy: 'cal.bin': No such file
avr32-gcc -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -Os -isystem /home/raman/Skullets/fuzix/FUZIX/Library/include -isystem /home/raman/Skullets/fuzix/FUZIX/Library/include/uc3a0512 -Wno-char-subscripts   -c -o chmem.o chmem.c
/home/raman/Skullets/fuzix/FUZIX/Library/link/lduc3a0512 /home/raman/Skullets/fuzix/FUZIX/Library/libs/crt0_uc3a0512.o chmem.o -o chmem -L/home/raman/Skullets/fuzix/FUZIX/Library/libs --gc-sections --emit-relocs -lcuc3a0512 -L/home/raman/tools/avr32-gnu-toolchain-linux_x86/bin/../lib/gcc/avr32/4.4.7/ -lgcc -T /home/raman/Skullets/fuzix/FUZIX/Library/elf2flt.ld 
avr32-gcc -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-builtin -Wall -Os -isystem /home/raman/Skullets/fuzix/FUZIX/Library/include -isystem /home/raman/Skullets/fuzix/FUZIX/Library/include/uc3a0512 -Wno-char-subscripts   -c -o cksum.o cksum.c
[... cut ...]
make: *** [Makefile:141: apps] Error 2
```

I also tried removing the files that complain from SRC within the
Makefile. The build is able to go through in that case.

What am I getting wrong? Could you please give me a hint?

Second: While I'm trying to get familiar with the FUZIX code-base, is
there something I can help with? For example, with documentation OR
relatively easy code fixes? I have an rpipico target with me to
reproduce @davidgiven's work. May I help with something there? Many
thanks.

ramangopalan, 2024-01-23:
> You should get to starting the shell before setjmp/longjmp burns
> you. The Bourne Shell uses it at startup and then for various kind
> of error and end of job handling

I fully understand Alan. Sure. I'll keep this in mind.

EtchedPixels, 2024-01-23:
avr32-ld: BFD (AVR_32_bit_GNU_Toolchain_3.4.2_435) 2.23.1 assertion fail

Your linker is broken. Unfortunately this seems to be a bit common with GNU linkers, especially forks of them. Nobody actually tests anything but the standard build options. That may need some detective work to see if it's a known problem and if there is a workaround, or a way to change the link file to dodge it.

Is your tree accessible anywhere as it's probably going to be easier to debug this with the tree merged here and a copy of the right tool chain.

ramangopalan, 2024-01-24:
> Your linker is broken. Unfortunately this seems to be a bit common
> with GNU linkers, especially forks of them. Nobody actually tests
> anything but the standard build options. That may need some [...]

Thank you Alan. I suspected the toolchain too but I wasn't very sure.
I understand.

> Is your tree accessible anywhere as it's probably going to be easier
> to debug this with the tree merged here and a copy of the right tool
> chain.

Sure. It is on the avr32 branch here [1]. [2] is the avr32-gcc I used
to check the build with FUZIX. [2] is able to build projects like
RT-Thread, eLua and NuttX for AVR32.

Please note: I still haven't imported the clock, USART and device
initialization parts code for devtty and devsdspi since the build
doesn't reach this point yet on GNU/Linux.

References:
[1]: https://github.com/ramangopalan/FUZIX/tree/avr32
[2]: https://github.com/ramangopalan/avr32-gnu-toolchain-linux_x86_64