EtchedPixels/FUZIX #1046: esp32 build

FransM, 2024-02-19:
I am trying to get FUZIX on my esp32.
As I could not find a mailing list or so I am writing down what I did here:
Build with the latest esp-idf tools did not work (I got errors on conflicting definitions of types like ssize_t).

Triggered by the commit that did set esp-idf for CI to 5.0.2 I used that version.
It now builds but when I run it it quickly runs in a  panic:

FUZIX version 0.5
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2024 Alan Cox <alan@etchedpixels.co.uk>
Devboot

panic: map_init

Tormod suggested me to try commit 4b85b66fefbcbf50002198b211467425321a6371
This version also  builds (after doing a make clean of course) but that one gives me an illegal instruction right away.
(it does not even print the FUZIX version or the Copyright messages.
Fatal exception (0): IllegalInstruction
epc1=0x4007600f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Anyone a suggestion on how to proceed?

Oh and a side note: I found that make clean does not clean all. I'll try to author a fix.

EtchedPixels, 2024-02-20:
@davidgiven

davidgiven, 2024-02-20:
Hello --- I haven't touched this for ages. I got as far as launching the kernel, but then it would crash partway through initialisation with symptoms that looked like a stack corruption. Working on the ESP32 is painful due to the windowed register system; the stack in memory is cached in invisible registers which aren't accessible to the user, and data is marshalled in and out of them via special exceptions, and getting these all exactly right is a painful experience. I see that there are now some decent simulators so it may be worth another look.

coolcoder613eb, 2024-04-21:
I am interested in this.

bialystok, 2024-04-29:
and wifi wpa3

closed 2024-06-16 by EtchedPixels

mix127, 2024-08-06:
What is status esp32 mechine?

davidgiven, 2024-08-07:
No motion, sorry. I would like to get back to it, but there are too many projects, too little time...

mix127, 2024-08-07:
ok, I undestand