EtchedPixels/FUZIX #1062: rpipico platform not working

Challmymind, 2024-04-07:
Newest pico build is not working. After flashing filesystem.uf2 then fuzix.uf2 there is no output on serial  (ttyACM0), sadly i cannot test if UART is working. I have done the same with the tarballs from [here](http://cowlark.com/2021-03-10-fuzix-pi-pico/index.html) with the success. Newest branch sets the LED on but as far as i can tell after it sits dormant. I will try to build version using previous commits before IOCTL changes (i dont know if it has broken things i just want to give it a try).

EtchedPixels, 2024-04-07:
I suspect @veremenko-y 's usb serial changes might be a more likely candidate

Does a682f7e38ae477c525b80670e7ea2d1127b3cd84 run as a starting point - that's just before all the recent changing around.

Challmymind, 2024-04-07:
You were right, a682f7e38ae477c525b80670e7ea2d1127b3cd84 builds correctly and everything works
I might try to revert changes to the USB and keep those from IOCTL, maybe it will also work (as it should) without any additional fixes.

Challmymind, 2024-04-07:
Update: shutdown is also invalid. 34ff65a1186027d45b63f88e443787799cff8695 is working until closed by shutdown command, after shutting it down it won't boot again.

veremenko-y, 2024-04-07:
Ugh... I'll look at it today. What's your set up? Just bare pico and USB serial?

veremenko-y, 2024-04-07:
> Update: shutdown is also invalid. [34ff65a](https://github.com/EtchedPixels/FUZIX/commit/34ff65a1186027d45b63f88e443787799cff8695) is working until closed by shutdown command, after shutting it down it won't boot again.

So I know what's happening with this one. Because you're using usb serial, it takes usually skips the starting messages. Even remount shutdown still corrupts file system for some reason. And after boot you get fsck prompt. If you type `n` or `y`, it'll continue booting. 

Left is serial, right is USB.
[image]

I had the same issue before my changes as well. If you add a sleep call after `core1_init();` in main.c, you'll likely see full boot.

What I'm going to do is:
* add a check to wait until USB is initialized, and only after that continue booting
* add config to switch boot tty to USB one

@EtchedPixels One question for you. Init refers to `/dev/tty1` explicitly. What would be an appropriate way to configure it? Would something like this pass review? :)

    // init.c:559
    do {
    #ifdef INIT_TTY
	    fdtty1 = open(INIT_TTY, O_RDWR|O_NOCTTY);
    #else
	    fdtty1 = open("/dev/tty1", O_RDWR|O_NOCTTY);
    #endif
    } while (fdtty1 < 0);

EtchedPixels, 2024-04-07:
tty1 is always the console

How devtty.c maps that minor is up to the platform

veremenko-y, 2024-04-10:
I think I have a fix to reconfigure TTYs in any order. I need to test it properly in hardware, but expect PR in a few days.

veremenko-y, 2024-04-15:
@Challmymind 
Are you able to test #1063 ?

fab1an2, 2024-04-29:
where is tutorial how run it?
I have rpi2040 but how start? what program to flash etc?

veremenko-y, 2024-04-29:
> where is tutorial how run it? I have rpi2040 but how start? what program to flash etc?
> 
> i use linux

This is unrelated to the current issue. Please create new one or post on discussion

veremenko-y, 2024-06-07:
@Challmymind please try again!

closed 2024-06-30 by EtchedPixels