EtchedPixels/FUZIX #1066: Building for v8080 fails

svofski, 2024-04-25:
I have built Fuzix-Compiler-Kit for 8080, it's in /opt/fpc.

In Fuzix tree I edited the Makefile to target v8080.

It starts out nicely, but then breaks here (see below).

It seems to not be able to find `lorder85`, which Fuzix-Compiler-Kit doesn't seem to build either. Can I use `lorder8080` in place of `lorder85`?

```
ar rc m8080.lib `lorder85 acosf.o acoshf.o asinf.o asinhf.o atan2f.o atanf.o atanhf.o cbrtf.o ceilf.o copysignf.o erff.o expf.o expm1f.o fabsf.o fdimf.o floorf.o fmaxf.o fminf.o fmodf.o frexpf.o hypotf.o ilogbf.o j0f.o j1f.o jnf.o ldexpf.o lgammaf.o lgammaf_r.o logf.o log2f.o log10f.o logbf.o lrintf.o lroundf.o modff.o nearbyintf.o nextafterf.o powf.o remainderf.o remquof.o rintf.o roundf.o scalbnf.o scalbln.o sinf.o sincosf.o sinhf.o sqrtf.o tgammaf.o vfscanf_m.o vfprintf_m.o __expo2f.o __float_bits.o __fpclassifyf.o __log1pf.o __signgam.o mathhelper.o | tsort`
/bin/sh: 1: lorder85: not found
ln -sf m8080.lib libm8080.a
cp libm8080.a libm.a
make[1]: Leaving directory '/home/svo/FUZIX/Library/libs'
make[1]: Entering directory '/home/svo/FUZIX/Library/libs'
make[1]: 'install' is up to date.
make[1]: Leaving directory '/home/svo/FUZIX/Library/libs'
(cd Applications; make)
make[1]: Entering directory '/home/svo/FUZIX/Applications'
(cd util; make -f Makefile.8080)
make[2]: Entering directory '/home/svo/FUZIX/Applications/util'
fcc -X -m8080 -Os -D__STDC__ -c -I/home/svo/FUZIX/Library/include -I/home/svo/FUZIX/Library/include/8080   -c -o chmem.o chmem.c
<command-line>: warning: "__STDC__" redefined
<built-in>: note: this is the location of the previous definition
/home/svo/FUZIX/Library/link/ld8080 /home/svo/FUZIX/Library/libs/crt0_8080.o chmem.o -o chmem -L/home/svo/FUZIX/Library/libs -lc8080
libpath '/opt/fcc//lib/8080/'
/home/svo/FUZIX/Library/libs/crt0_8080.o: Unknown symbol '___stdio_init_va'.
/home/svo/FUZIX/Library/libs/crt0_8080.o: Unknown symbol '___argv'.
/home/svo/FUZIX/Library/libs/crt0_8080.o: Unknown symbol '_exit'.
chmem.o: Unknown symbol '_printf'.
chmem.o: Unknown symbol '_printf'.
chmem.o: Unknown symbol '_sscanf'.
chmem.o: Unknown symbol '_stderr'.
chmem.o: Unknown symbol '_fprintf'.
chmem.o: Unknown symbol '_exit'.
chmem.o: Unknown symbol '_rewind'.
chmem.o: Unknown symbol '_fwrite'.
chmem.o: Unknown symbol '_stderr'.
chmem.o: Unknown symbol '_fprintf'.
chmem.o: Unknown symbol '_exit'.
chmem.o: Unknown symbol '_stderr'.
chmem.o: Unknown symbol '_fprintf'.
chmem.o: Unknown symbol '_exit'.
chmem.o: Unknown symbol '___fopen'.
chmem.o: Unknown symbol '___fopen'.
chmem.o: Unknown symbol '_perror'.
chmem.o: Unknown symbol '_exit'.
chmem.o: Unknown symbol '_fread'.
chmem.o: Unknown symbol '_stderr'.
chmem.o: Unknown symbol '_fprintf'.
chmem.o: Unknown symbol '_exit'.
chmem.o: Unknown symbol '_stderr'.
chmem.o: Unknown symbol '_fprintf'.
chmem.o: Unknown symbol '_exit'.
chmem.o: Unknown symbol '_fclose'.
chmem.o: Unknown symbol '_exit'.
make[2]: *** [Makefile.common:130: chmem] Error 1
make[2]: Leaving directory '/home/svo/FUZIX/Applications/util'
make[1]: *** [Makefile:11: util] Error 2
make[1]: Leaving directory '/home/svo/FUZIX/Applications'
make: *** [Makefile:137: apps] Error 2
```

svofski, 2024-04-25:
Update: I linked lorder8080 to lorder85 to see where it goes. It went much farther but still couldn't finish, these were the final lines:

```
(skipped a lot of similar warnings) 

<command-line>: warning: "__STDC__" redefined
<built-in>: note: this is the location of the previous definition
/home/svo/FUZIX/Library/link/ld8080 /home/svo/FUZIX/Library/libs/crt0nostdio_8080.o while1.o -o while1 -L/home/svo/FUZIX/Library/libs -lc8080
libpath '/opt/fcc//lib/8080/'
fcc -X -m8080 -Os -D__STDC__ -c -I/home/svo/FUZIX/Library/include -I/home/svo/FUZIX/Library/include/8080   -c -o whoami.o whoami.c
<command-line>: warning: "__STDC__" redefined
<built-in>: note: this is the location of the previous definition
/home/svo/FUZIX/Library/link/ld8080 /home/svo/FUZIX/Library/libs/crt0nostdio_8080.o whoami.o -o whoami -L/home/svo/FUZIX/Library/libs -lc8080
libpath '/opt/fcc//lib/8080/'
fcc -X -m8080 -Os -D__STDC__ -c -I/home/svo/FUZIX/Library/include -I/home/svo/FUZIX/Library/include/8080   -c -o yes.o yes.c
<command-line>: warning: "__STDC__" redefined
<built-in>: note: this is the location of the previous definition
/home/svo/FUZIX/Library/link/ld8080 /home/svo/FUZIX/Library/libs/crt0nostdio_8080.o yes.o -o yes -L/home/svo/FUZIX/Library/libs -lc8080
libpath '/opt/fcc//lib/8080/'
fcc -X -m8080 -Os -D__STDC__ -c -I/home/svo/FUZIX/Library/include -I/home/svo/FUZIX/Library/include/8080   -c -o fleamacs.o fleamacs.c
<command-line>: warning: "__STDC__" redefined
<built-in>: note: this is the location of the previous definition
/home/svo/FUZIX/Library/link/ld8080 /home/svo/FUZIX/Library/libs/crt0_8080.o fleamacs.o -o fleamacs -L/home/svo/FUZIX/Library/libs -ltermcap8080 -lc8080
libpath '/opt/fcc//lib/8080/'
fleamacs.o: Unknown symbol '_tputs'.
fleamacs.o: Unknown symbol '_tgoto'.
make[2]: *** [Makefile.common:136: fleamacs] Error 1
make[2]: Leaving directory '/home/svo/FUZIX/Applications/util'
make[1]: *** [Makefile:11: util] Error 2
make[1]: Leaving directory '/home/svo/FUZIX/Applications'
make: *** [Makefile:137: apps] Error 2
```

svofski, 2024-04-25:
I see there were some changes in the codebase so I tried building again from scratch, this time this is what I've got:

```
ar rc m8080.lib `lorder8080 acosf.o acoshf.o asinf.o asinhf.o atan2f.o atanf.o atanhf.o cbrtf.o ceilf.o copysignf.o erff.o expf.o expm1f.o fabsf.o fdimf.o floorf.o fmaxf.o fminf.o fmodf.o frexpf.o hypotf.o ilogbf.o j0f.o j1f.o jnf.o ldexpf.o lgammaf.o lgammaf_r.o logf.o log2f.o log10f.o logbf.o lrintf.o lroundf.o modff.o nearbyintf.o nextafterf.o powf.o remainderf.o remquof.o rintf.o roundf.o scalbnf.o scalbln.o sinf.o sincosf.o sinhf.o sqrtf.o tgammaf.o vfscanf_m.o vfprintf_m.o __expo2f.o __float_bits.o __fpclassifyf.o __log1pf.o __signgam.o mathhelper.o | tsort`
ln -sf m8080.lib libm8080.a
cp libm8080.a libm.a
make[1]: Leaving directory '/home/svo/fuzix/FUZIX/Library/libs'
make[1]: Entering directory '/home/svo/fuzix/FUZIX/Library/libs'
make[1]: 'install' is up to date.
make[1]: Leaving directory '/home/svo/fuzix/FUZIX/Library/libs'
(cd Applications; make)
make[1]: Entering directory '/home/svo/fuzix/FUZIX/Applications'
(cd util; make -f Makefile.8080)
make[2]: Entering directory '/home/svo/fuzix/FUZIX/Applications/util'
fcc -X -m8080 -Os -D__STDC__ -c -I/home/svo/fuzix/FUZIX/Library/include -I/home/svo/fuzix/FUZIX/Library/include/8080   -c -o fsck-fuzix.o fsck-fuzix.c
<command-line>: warning: "__STDC__" redefined
<built-in>: note: this is the location of the previous definition
/home/svo/fuzix/FUZIX/Library/link/ld8080 /home/svo/fuzix/FUZIX/Library/libs/crt0_8080.o fsck-fuzix.o -o fsck-fuzix -L/home/svo/fuzix/FUZIX/Library/libs -lc8080
libpath '/opt/fcc//lib/8080/'
fcc -X -m8080 -Os -D__STDC__ -c -I/home/svo/fuzix/FUZIX/Library/include -I/home/svo/fuzix/FUZIX/Library/include/8080   -c -o ps.o ps.c
<command-line>: warning: "__STDC__" redefined
<built-in>: note: this is the location of the previous definition
/home/svo/fuzix/FUZIX/Library/link/ld8080 /home/svo/fuzix/FUZIX/Library/libs/crt0_8080.o ps.o -o ps -L/home/svo/fuzix/FUZIX/Library/libs -lc8080
libpath '/opt/fcc//lib/8080/'
fcc -X -m8080 -Os -D__STDC__ -c -I/home/svo/fuzix/FUZIX/Library/include -I/home/svo/fuzix/FUZIX/Library/include/8080   -c -o tget.o tget.c
<command-line>: warning: "__STDC__" redefined
<built-in>: note: this is the location of the previous definition
/home/svo/fuzix/FUZIX/Library/link/ld8080 /home/svo/fuzix/FUZIX/Library/libs/crt0_8080.o tget.o -o tget -L/home/svo/fuzix/FUZIX/Library/libs -ltermcap8080 -lc8080
libpath '/opt/fcc//lib/8080/'
tget.o: Unknown symbol '_tgetent'.
tget.o: Unknown symbol '_tgetflag'.
tget.o: Unknown symbol '_tgetnum'.
tget.o: Unknown symbol '_tgetstr'.
tget.o: Unknown symbol '_tputs'.
tget.o: Unknown symbol '_tgetstr'.
tget.o: Unknown symbol '_tgoto'.
tget.o: Unknown symbol '_tputs'.
make[2]: *** [Makefile.common:136: tget] Error 1
make[2]: Leaving directory '/home/svo/fuzix/FUZIX/Applications/util'
make[1]: *** [Makefile:11: util] Error 2
make[1]: Leaving directory '/home/svo/fuzix/FUZIX/Applications'
make: *** [Makefile:137: apps] Error 2
```

EtchedPixels, 2024-04-25:
What is there in the log for the build of libtermcap, because I can't reproduce that here at this point

svofski, 2024-04-25:
It's strange because I seem to end up with a different error when I run make multiple times.

I made some quick and dirty modifications to Applications/CC/support8080/Makefile, this allowed me to advance further
```
diff --git a/Applications/CC/support8080/Makefile b/Applications/CC/support8080/Makefile
index 509797304..96a10386d 100644
--- a/Applications/CC/support8080/Makefile
+++ b/Applications/CC/support8080/Makefile
@@ -49,17 +49,18 @@ include ldst.mk

 ldword/_10.o: makeldst
        ./makeldst
-       cc85 -c ldword/_10.s
+       fcc -DNO_FLOAT -X -m8080 -Os -D__STDC__ -c -I/home/svo/fuzix/FUZIX/Library/include -I/home/svo/fuzix/FUZIX/Library/include/8080 -DCPU_8080 -c ldword/_10.s

 makeldst: makeldst.c
        $(CC) makeldst.c -o ./makeldst
-
+       mkdir -p ldword stword ldbyte stbyte
 .s.o:
-       cc85 -c $<
+       fcc -DNO_FLOAT -X -m8080 -Os -D__STDC__ -c -I/home/svo/fuzix/FUZIX/Library/include -I/home/svo/fuzix/FUZIX/Library/include/8080 -DCPU_8080  -c $<
+       #cc85 -c $<

 lib8080.a: makeldst $(OBJ)
        rm -f lib8080.a
-       ar qc lib8080.a `../lorder85 $(OBJ) | tsort`
+       ar qc lib8080.a `../lorder$(PLATBIN) $(OBJ) | tsort`

 clean:
        rm -f *.o *.a
```

If this passes without trouble for you, could it be that your build env has both fcc and cc85 at the same time?

EtchedPixels, 2024-04-26:
I really thought I'd cleaned all the old ones out but apparently not.

Cleaned that up, fixed an 8085 library location and got v8080 rcbus-8080 and rcbus-8085 building with just the new stuff on the path. Fingers crossed 8)

svofski, 2024-04-26:
Thanks! I hope I'm not too much of a nuisance :)

BTW, what is this v8080 platform, I couldn't find any links to the emulator? The files in the downloads on fuzix.org look like they're made for z80pack, but I couldn't load v8080 version in z80pack (z80pack version works fine in z80pack). I'm mildly curious about possibly porting Fuzix to Vector-06c, a real 8080 based platform, but I probably can't lift that without a working reference.

svofski, 2024-04-26:
This is necessary, or it fails to create the _*.s files
```
diff --git a/Applications/CC/support8080/Makefile b/Applications/CC/support8080/Makefile
index ffd00b57c..5c4df490b 100644
--- a/Applications/CC/support8080/Makefile
+++ b/Applications/CC/support8080/Makefile
@@ -48,6 +48,7 @@ OBJ = workspace.o __true.o __switchc.o __switch.o __switchl.o __pushl.o __sex.o
 include ldst.mk

 ldword/_10.o: makeldst
+       mkdir -p ldword stword ldbyte stbyte
        ./makeldst
        fcc -m8080 -c ldword/_10.s
```

svofski, 2024-04-26:
Next:

```
....
fcc -m8080 -c stbyte/_8.s
fcc -m8080 -c stbyte/_9.s
rm -f lib8080.a
ar qc lib8080.a `../lorder85 workspace.o __true.o __switchc.o __switch.o __switchl.o __pushl.o __sex.o __ldwordw.o __and.o __andeq.o __or.o __oreq.o __xor.o __xoreq.o __andeqde.o __oreqde.o __xoreqde.o __minuseq.o __pluseq.o __pluseqn.o __minuseqn.o __pluseqde.o __minuseqde.o __diveq.o __divequ.o __modeq.o __modequ.o __muleq.o __bool.o __cpl.o __not.o __boolc.o __cplc.o __notc.o __minus.o __plus.o __postinc.o __postdec.o __postincn.o __postdecn.o __shr.o __shleq.o __shrequ.o __shreq.o __shlde.o __divdeu.o __divde.o __cclt.o __ccltu.o __ccgteq.o __ccgtequ.o __cceq.o __ccne.o __cmpeq.o __cmpne.o __cmpeqb.o __cmpneb.o __ccgt.o __ccgtu.o __cclteq.o __ccltequ.o __cmpeq0.o __cmpgtu.o __cmpltu.o __cmpltequ.o __cmpgtequ.o __cmpgt.o __cmplt.o __cmplteq.o __cmpgteq.o __cmpgt0.o __cmplteq0.o __cmpgteq0.o __cmplt0.o __andc.o __orc.o __xorc.o __andeqc.o __oreqc.o __xoreqc.o __minuseqc.o __pluseqc.o __modeqc.o __modequc.o __diveqc.o __divequc.o __castc.o __castcl.o __castu_l.o __cast_l.o __cast_ul.o __postincc.o __postdecc.o __shrdeuc.o __shrequc.o __shleqc.o __shreqc.o __shrequc.o __bandl.o __borl.o __bxorl.o __oreql.o __andeql.o __xoreql.o __pluseql.o __minuseql.o __booll.o __cpll.o __notl.o __negatel.o __cmpl.o __ccltl.o __ccgteql.o __ccgtl.o __cclteql.o __ccltul.o __ccgtequl.o __ccgtul.o __ccltequl.o __ccequl.o __ccneul.o __derefl.o __assignl.o __minusl.o __plusl.o __mull.o __divl.o __postincl.o __postdecl.o __shll.o __shrl.o __shrul.o __shleql.o __shreql.o __bandde.o __borde.o __bxorde.o __mulde.o __shrdeu.o __muldec.o __ldword.o __ldwordw.o __stword.o __stwordw.o __ldbyte.o __ldbytew.o __stbyte.o __stbytew.o __bcand.o __bcdiv.o __bcdivu.o __bcmul.o __bcor.o __bcshl.o __bcshr.o __bcsub.o __bcxor.o _memcpy.o _memset.o _strlen.o ldword/_10.o ldword/_11.o ldword/_12.o ldword/_13.o ldword/_14.o ldword/_15.o ldword/_16.o ldword/_17.o ldword/_18.o ldword/_19.o ldword/_1.o ldword/_20.o ldword/_21.o ldword/_22.o ldword/_23.o ldword/_24.o ldword/_25.o ldword/_26.o ldword/_27.o ldword/_28.o ldword/_29.o ldword/_2.o ldword/_30.o ldword/_31.o ldword/_3.o ldword/_4.o ldword/_5.o ldword/_6.o ldword/_7.o ldword/_8.o ldword/_9.o ldbyte/_10.o ldbyte/_11.o ldbyte/_12.o ldbyte/_13.o ldbyte/_14.o ldbyte/_15.o ldbyte/_16.o ldbyte/_17.o ldbyte/_18.o ldbyte/_19.o ldbyte/_1.o ldbyte/_20.o ldbyte/_21.o ldbyte/_22.o ldbyte/_23.o ldbyte/_24.o ldbyte/_25.o ldbyte/_26.o ldbyte/_27.o ldbyte/_28.o ldbyte/_29.o ldbyte/_2.o ldbyte/_30.o ldbyte/_31.o ldbyte/_3.o ldbyte/_4.o ldbyte/_5.o ldbyte/_6.o ldbyte/_7.o ldbyte/_8.o ldbyte/_9.o stword/_10.o stword/_11.o stword/_12.o stword/_13.o stword/_14.o stword/_15.o stword/_16.o stword/_17.o stword/_18.o stword/_19.o stword/_1.o stword/_20.o stword/_21.o stword/_22.o stword/_23.o stword/_24.o stword/_25.o stword/_26.o stword/_27.o stword/_28.o stword/_29.o stword/_2.o stword/_30.o stword/_31.o stword/_3.o stword/_4.o stword/_5.o stword/_6.o stword/_7.o stword/_8.o stword/_9.o stbyte/_10.o stbyte/_11.o stbyte/_12.o stbyte/_13.o stbyte/_14.o stbyte/_15.o stbyte/_16.o stbyte/_17.o stbyte/_18.o stbyte/_19.o stbyte/_1.o stbyte/_20.o stbyte/_21.o stbyte/_22.o stbyte/_23.o stbyte/_24.o stbyte/_25.o stbyte/_26.o stbyte/_27.o stbyte/_28.o stbyte/_29.o stbyte/_2.o stbyte/_30.o stbyte/_31.o stbyte/_3.o stbyte/_4.o stbyte/_5.o stbyte/_6.o stbyte/_7.o stbyte/_8.o stbyte/_9.o | tsort`
fcc -m8080 -c crt0.s
make[3]: Leaving directory '/home/svo/fuzix/FUZIX/Applications/CC/support8080'
cp support8080/lib8080.a libcpu.a
make[2]: Leaving directory '/home/svo/fuzix/FUZIX/Applications/CC'
(cd cpp; make -f Makefile.8080)
make[2]: Entering directory '/home/svo/fuzix/FUZIX/Applications/cpp'
make[2]: Makefile.8080: No such file or directory
make[2]: *** No rule to make target 'Makefile.8080'.  Stop.
make[2]: Leaving directory '/home/svo/fuzix/FUZIX/Applications/cpp'
make[1]: *** [Makefile:77: cpp] Error 2
make[1]: Leaving directory '/home/svo/fuzix/FUZIX/Applications'
make: *** [Makefile:137: apps] Error 2
```

Easy fix:

```
cp Applications/cpp/Makefile.8085 Applications/cpp/Makefile.8080
```

And with this I could make it to the end!

svofski, 2024-04-26:
That's where it's at for now -- something is amiss.

```
Release 1.37, Copyright (C) 1987-2021 by Udo Munk

CPU speed is unlimited, CPU executes undocumented instructions

Booting...

B:..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
..........................
G
FUZIX version 0.5
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2024 Alan Cox <alan@etchedpixels.co.uk>
Devboot
400KiB total RAM, 336KiB available to processes (16 processes max)
Enabling interrupts ... ok.
bootdev: 1
Mounting root fs (root_dev=1, ro): failed
bootdev: 2
Mounting root fs (root_dev=2, ro): failed
bootdev: 0
Mounting root fs (root_dev=0, ro): OK
Starting /init
21 buffers added
init version 0.9.1
/etc/rc: : cannot execute
/etc/rc: : cannot execute
/etc/rc: : cannot execute
/etc/rc: : cannot execute
/etc/rc: /etc/mtab: cannot create
/etc/rc: /var/run/utmp: cannot create

 ^ ^
 n n   Fuzix 0.5
 >@<
       Welcome to Fuzix
 m m

login: root
login: unable to change owner of controlling tty

Welcome to FUZIX.
# ls /
: cannot execute
#
```

EtchedPixels, 2024-04-26:
Ok that's a miscompile that's crept in when I added register argument support although the actual bug was merely hidden before. It's miscompiling

```
void makearg(register char *args)
{
        ((ARGPTR) args)->argnxt = gchain;
        gchain = (ARGPTR) args;
}
```

on 8080 because for some reason it's optimizing the store to a cast of BC into LDAX B which is of course wrong for a 2 byte value. Need to dig into that a bit more.

EtchedPixels, 2024-04-26:
Ok should be fixed. You'll need to rebuild the userspace (/bin/sh anyway) after the compiler.

svofski, 2024-04-26:
Booted and works. Thanks!

svofski, 2024-04-27:
Where would you recommend picking up if I'm to port it to v06c? The system is like so:

No ROM

0000-7FFF unmapped RAM
8000-FFFF 4x 8K bitplanes in the main area which can be mapped for CPU access

Extra 256K as 4x 64K pages. 
For each 64K page, the top 32K can be switched on to the main RAM, shadowing screen area. 
The bottom 32K is only accessible as stack.

I browsed the platforms but didn't find one that would be sufficiently similar to this yet.

EtchedPixels, 2024-04-27:
How does the "only accessible as stack" work ?

In general though I would probably start with v8080 and change the memory map so it's
0000-00FF RST vectors etc as before
0100 up Common then data, bss, buffers, code
8000-FFFF Switching between video / some of the kernel / user processes

That will require some changes as the 8080 port currently doesn't have relocatable user space binaries because I had no target needing them and I was concerned about how slow the relocations would be on an 8080. Linking the user space at 0x8000 would get you started though.

svofski, 2024-04-27:
So we put the kernel and buffers in the lower addresses, I see.

> How does the "only accessible as stack" work ?
8080 has STACK status bit (bit 2 in the status word that it outputs on SYNC). It is used to select a different memory bank. A typical transfer would be: disable interrupts, save SP, set it up for transfer, push or pop your data, restore SP, enable interrupts.

EtchedPixels, 2024-04-27:
So CALL RET PUSH POP but nothing else. I guess that means the lower 32K is only really usable as a ramdisc ?

svofski, 2024-04-27:
It's controlled by bits in an I/O port, normally this feature is switched off and the lower 32K are just normal 32K.

commit eeb8ac5e (2024-04-26): 8080/5: Fix build problems where it assumed the old cc85 was also present

Closes #1066
closed 2024-05-30 by EtchedPixels