EtchedPixels/FUZIX #1085: TMS9918 Video not working on platform-2063

dennowiggle, 2024-09-14:
I tested FUZIX on [2063-Z80 Retro!](https://github.com/Z80-Retro/2063-Z80) with [2068-Z80-TMS9118](https://github.com/Z80-Retro/2068-Z80-TMS9118) using the pre-built image from the website and a freshly compiled 4.0 version. I have a blue screen with garbled characters.
[image]

I tried replacing 2xnop instructions with `ex (sp),hl` but the screen is still garbled. When I hit enter I can see the screen respond so it's like the screen data is okay but the font has not been loaded to the right place, or the font address area is not setup correctly on the TMS.

I don't mind helping to debug but I'm in the dark about how to proceed. Do we know that `Kernel/dev/vdp1.s` is working okay on real systems? If yes, which system so that can be used as a model for comparison?

I took a look at rc2014 but that uses it's own vdp1.s and the VDP seems to be setup with c code. Maybe that is the way to go?

dennowiggle, 2024-10-08:
I added extra VDP_DELAY statements throughout the dev/vdp1.s file and then the video worked. No change was needed to the 3 x `nop` in the platform vdp.s file. 

This was tested on the 0.4 version of FUZIX on two 2063 Z80 Retro! systems with TMS9118. File attached. 

I was not able to optimise the number of VDP_DELAY statements easily and I am not sure how to test the rop and wop code.
[vdp1.zip](https://github.com/user-attachments/files/17297180/vdp1.zip)

EtchedPixels, 2024-11-12:
Think I found the one that was wrong. When restoring the font one of the delays was the wrong side of the out()

closed 2024-12-05 by EtchedPixels